ShikOfTheRa/scarab-osd #297: OSD turns off after arming - noisy power to the OSD

eluminer, 2016-08-17:
Hello!

I have my 210 quad set up like this:

RS2205 2300 KV
LittleBee 20A
X-Racer F303 V2.1
David Windestals VR (linear VR to 12V + LC filter)
APM Power Module for current measurement (to FC) and **5V power**
Micro MinimOSD powered from UART2 from FC
4S pack

_**Problem:**_
After arming OSD, or even entire video feed, disappears. Worse on higher throttle and fast pitch/roll movement. Better on 3S pack. Even while idling OSD disappears and video starts to be noisy.

_**Notes:**_
No ground loops and everything is grounded (camera, VTX, OSD, FC)

**_Tried solutions:_**
1. First I tried connecting camera to VTX directly. Perfect link, no noise, due to 12V LC filter doing it's job well.
2. Put 1000 uF 35 V capacitor on 5V pins on OSD. Now video never disappears but OSD overlay does. - slightly better, but not a solution.
3. Put inductance on 5V OSD in - no change
4. Used separate 5V BEC (maybe APM power module is bad) - no change.
5. Used separate 5V BEC but as an input I used 12 V filtered out for the camera (so, clean power) as opposed to 4S voltage from the pack (unfiltered) as in case #4 - no change.
6. Put another 1000 uF 35 V capacitor on 5V rail on FC (basically where the 5V input is) - no change.

All in all, capacitor on 5V OSD pins helps (even 220 uF), but it's not enough. Can't make it flyable.

My conclusion is that the problem is 5V power to the OSD, but can't find a way to make this better.

Any suggestions.

P.S.
Test are done on the bench with props off. I even put 1 prop on and it's the same. I think that it should be good on the bench before I try it with the props on on higher currents.

DzikuVx, 2016-08-17:
I would say it is hardware, not software issue

jflyper, 2016-08-17:
> Problem:
Even while idling OSD disappears and video starts to be noisy.

Never heard of OSD disappearance this bad.

> 5. Used separate 5V BEC but as an input I used 12 V filtered out for the camera (so, clean power) as opposed to 4S voltage from the pack (unfiltered) as in case #4 - no change

This is really weird. Do you still loose overlay on idle with this measure?

As @DzikuVx pointed out, I would try another OSD, or at least restart everything from desktop experiment in which only CAM, OSD and VTX are connected to stable power.

devosthomas, 2016-08-18:
Hi Elumier,

I have the same issue also with 
RS2205 2300 KV
LittleBee 20A
SProRacing F3 FC
Micro minim OSD

I have read somewhere that replacing the MicroMinim OSD would be the solution:
http://www.rcgroups.com/forums/showthread.php?t=2623451 (see last message)

eluminer, 2016-08-18:
Thanks for the help so far guys.

Here are the news:
I've changed the grounding point of the osd to the battery negative lead directly. Helped a little bit, not to much.
Then I've put 470uF 25V capacitor on each LittleBee 20A ESC and that helped a lot. Still not 100% because with props on and with extreme changes I still get noise in the video. However, I have 2 spares OSD so I will try to change it right now.
Seems like ceramic capacitors on LittleBees just don't do their job. Problem is that I can't fit 470 uF int the frame since they are too big. :-( The biggest one is 100 uF that I can fit, but this one doesn't do the job.

Will keep you posted.

Best regards!

eluminer, 2016-08-18:
Brand new Micro MinimOSD - no difference. OSD was/is fine.

jflyper, 2016-08-18:
@eluminer Did you try Low ESR capacitors?

[URL="https://www.youtube.com/watch?v=ZhvuQJPGHzU"]https://www.youtube.com/watch?v=ZhvuQJPGHzU[/URL]

eluminer, 2016-08-19:
@jflyper 
Well, 100 uF are not low ESR. Thanks for the link, it is great and exactly what I needed. It seems loe ESR is very important. I think I will rewire everything and use voltage regulator with 1000 uF low ESR capacitor. I might add some 220 uF to each ESC just because I have space for that, but definitely 1000 uF on main leads. Seems it does the job well.

Will report back!

Tomcatt, 2016-08-24:
I think you guys may want to look into a good PDB.  That way it's filtered and you a good 5V ref. for the OSD and 12V for the video an transmitter.

If you have an extra 12V line, send that to the camera or if your video transmitter supports, use the power from it and send that to the camera.

Also use ALL the grounds too.  Better safe than sorry.

cpusmith, 2016-08-24:
I was having all those problem that you described until I discovered this:

http://www.banggood.com/CC3D-NAZE32-F3-Power-Distribution-Board-PDB-With-Filter-BEC-Output-5V-12V-3A-for-QAV250-p-1023543.html

Now it's perfect.  I do still put a capacitor on the 5V input of the microminumOSD though to prevent brownouts.

I have 6 quads and discovered this 2 quads ago and went back and replaced the ESC and BEC power - now it's great with the added bonus of cleaner wiring.

You can even get one with the OSD built in: http://www.banggood.com/MXK-Power-Distribution-Board-PDB-with-OSD-BEC-Output-5V-12V-3A-LC-Filter-for-CC3C-Naze32-F3-p-1031375.html

**Board Layout**
http://img.banggood.com/images/oaupload/banggood/images/73/9B/203C101057C38938B2237EF2895ACF92.jpg

**Connectivity Diagram**
https://ae01.alicdn.com/kf/HTB1u54ALVXXXXcIXFXXq6xXFXXXL/205920023/HTB1u54ALVXXXXcIXFXXq6xXFXXXL.jpg?size=148526&height=561&width=757&hash=97752e4bb20e6a5e2573770d5c86f528

Even cheaper on Aliexpress: https://www.aliexpress.com/item/2-6S-Power-distribution-board-5V-12V-Dual-BEC-LC-filter-3-in-1-PCB-PDB/32617991613.html?spm=2114.13010608.0.94.6aTwN6

Tomcatt, 2016-08-24:
This I have in 2 of mine.

http://www.aliexpress.com/item/New-Matek-V3-5-in-1-Power-Distribution-Board-PDB-Hub-With-Dual-BEC-5V-12V/32658114724.html?spm=2114.01010208.3.43.fo3ww4&ws_ab_test=searchweb201556_0,searchweb201602_4_10057_10056_10065_10055_10054_10059_10058_10017_107_10060_10061_10052_414_10062_10053_10050_10051,searchweb201603_4&btsid=2ded63bf-71f4-47de-9061-026287fe2a38

closed 2016-09-14 by ShikOfTheRa