ShikOfTheRa/scarab-osd #299: Occasional resets of settings

sebastianha, 2016-08-21:
Hi,

from time to time I experience a reset of all my settings and the OSD shows the default settings (at least it looks like this).

I already put the hardware in tape so nothing could touch the hardware. I am using the Micro MinimOSD (http://www.ebay.de/itm/222086270087) with a ZMR250 frame with integrated PDB (http://www.ebay.de/itm/262492252754). I am running Cleanflight 1.12.1 on a Naze32 V6.

I could not determine what causes the reset yet, do you have any idea? Sometime it works for a week or two, sometimes only for a few days.

jflyper, 2016-08-22:
Is it only the settings that gets wiped? How about the firmware itself? Do you have to re-flash (upload) your sketch, too?

sebastianha, 2016-08-22:
Hi,

the OSD software itself is still available so I think only the settings are wiped. I can use the GUI to set everything again. I do not need to flash the sketch again.

1) https://github.com/ShikOfTheRa/scarab-osd/commit/a9f15375c866137f84603167d77c1ca0a56ccd70
2) No, no problems at all.

ShikOfTheRa, 2016-08-22:
This is not something I've experienced either. My guess would be same as jf - power spike related. Or less likely a dodgy chip. I think if it was a SW issue we would have had it raised previously

sebastianha, 2016-08-22:
Ok, thank you very much. What exactly is the problem with power spikes? Which effect does clear the EEPROM (settings are saved in the EEPROM I guess).

ShikOfTheRa, 2016-08-22:
Its just a guess at the most likely cause - spikes and brownouts are documented to cause issues with AVR and probably many other chips too, but as you are not getting any OSD resets, this makes this a lot less likely.
Issue is mainly high speed switching and current draw from the ESC.

If it were me: 
1 Check ground is good
2 Next  I'd try a big cap on the 5v supply to the OSD.
3 If not fixed I'd swap OSD

sebastianha, 2016-08-22:
Thanks for the suggestions, I will try it and will let you know if anything helps.

sebastianha, 2016-08-22:
Do you know what the voltage input range is for the Micro MinimOSD? Is it exactly 5V?

When I remember correctly I have measured slightly less on my regulated voltage, perhaps about 4.6V. But the Atmega (where the EEPROM is located?!) is rated from 1.8–5.5V.

ShikOfTheRa, 2016-08-22:
I suspect manufacturer IC specs. 
4.6 I would expect to see some issue with max chip... that's more critical on voltage drops usually.
You wouldn't see the spike issues with a voltmeter. Would need a scope to see. Certainly exceeding those limits for some of the guys.
Normally the issues are highlighted by blackouts of OSD info- a real indicator. Hence jfp asking.
Never rule out a SW issue, but I think its low on probability scale.

sebastianha, 2016-08-22:
Ok, as I did not had any problems during flight my current guess is that somehow the EEPROM gets damaged during flight.

I found some information about EEPROM problems:
http://electronics.stackexchange.com/questions/77695/minimum-voltage-to-avoid-eeprom-corruption-for-atmega328p
http://atmel.force.com/support/articles/en_US/FAQ/Prevent-EEPROM-corruption
http://www.avrfreaks.net/forum/eeprom-data-corruption-1

One thing could also be wrong brown-out setting of the Atmega, I will check this, too.

jflyper, 2016-08-22:
For EEPROM to revert to default, checkEEPROM(), which is called during initialization must detect corrupted EEPROM and write default values. So, it is most likely that we are looking at EEPROM corruption problem, rather than reverting to default problem.

I suspect a bad or damaged ATmega328P chip.

While it may be possible to track your problem in depth, it is time consuming, and I can imagine you replacing the Micro anyway at the end. Get a new Micro from a different source if possible, and replace the current one. If the problem develops again, then you can start looking at rest of your system.

Side note:
> When I remember correctly I have measured slightly less on my regulated voltage, perhaps about 4.6V. But the Atmega (where the EEPROM is located?!) is rated from 1.8–5.5V.

The "safe operating range" is much narrower; for 16MHz operation, it's more like 4V3-5V5 or so. Googling around with "avr eeprom corruption" will give us some more details.

sebastianha, 2016-08-22:
Thanks for all the help, I already ordered a new Micro from another source and will let you know!

sebastianha, 2016-08-23:
Last night I added a 470uf 16v capacitor to my OSD, we will see if this will help.

jflyper, 2016-08-24:
It's extremely rare compared to other cases, but I hear & see people loosing their settings from time to time. I tend to ignore them because often times they are not associated with enough information, nor reproducible. But your case is different that you have a reproducible case. In that sense, I'm very interested in your case; bad M328 or power supply problem. Please share your findings with us!

sebastianha, 2016-08-24:
I still cannot force to reproduce the bug, but know I have a capacitor soldered to the power supply and I will continue watching it. The last days without flying it has kept its settings since I soldered the cap to it. In the next days I will fly again.

I will let you know when I find something!

sebastianha, 2016-08-28:
Status update: I did some flights and daily testing of the OSD settings and since I soldered the capacitor to it I had no reset of settings. I will continue testing.

jflyper, 2016-08-28:
Thanks for the update.

Hope the 470 actually works.

ShikOfTheRa, 2016-09-14:
@sebastianha how is it looking so far? Any improvement

sebastianha, 2016-09-14:
Hi, sorry,  forgot to update: Still everything is working fine. I had no reset since I added the capacitor!

jflyper, 2016-09-14:
Good! Happy flying!

closed 2016-09-14 by sebastianha