ShikOfTheRa/scarab-osd #300: Baro altitude and GPS differs althought there is no baro sensor (iNAV + CC3D)

sppnk, 2016-08-24:
When using iNAV  and CC3D (this FC has no barometer) I see in some flights the value of barometer altitude different (in more that 15 meters) than the value of the GPS altitude, althought I am using CC3D with this setup and there is no baro.

I have not adressed yet the circumstances when having this issue, and I guess (but not fully sure) that when using the same osd in other plane with naze (iNav and has baro), the values are also diferent, but they should be, as the GPS altitude precission is never the same as BARO altitude.

jflyper, 2016-08-24:
@sppnk Few questions
Are you FIXEDWING ?
Have you tweaked other options?
Why do you want to "Display BARO ALT" if you have no BARO?

ShikOfTheRa, 2016-08-24:
Default altitude displayed is whatever is sent BARO altitude from FC. If no BARO on FC, it possibly send random crap!
(Unless using fixed wing where BARO alt is replaced by GPS alt.)

So assuming not usingfixedwing, if no BARO can either:
Add this into config.h  
#define USEGPSALTITUDE
Or just turn of BARO alt and use GPS alt.

If using fixedwing with iNAV:
#define FIXEDWING

sppnk, 2016-08-24:
Yes, FIXEDWING. It is defined FIXEDWING already.

I was displaying before both with NAZE just to test the differences. Now with CC3D I guess it should show the same value in both places. Because I change osds bettween planes I didnt want to touch configurations each time I do that.

 Thank you, if this is not a bug I close it.

closed 2016-08-24 by sppnk

jflyper, 2016-08-25:
Hmmm...

My understanding was that MWOSD will not poll MSP_ALTITUDE (the BARO value) unless the BARO is set in available sensors (MSP_STATUS), nor try to display it.
While it can be fixed pragmatically by USEGPSALTITUDE, I wonder what's going on...

sppnk, 2016-08-25:
So, if iNAV is not sendind the correct MSP_STATUS, the problem is there. ?¿

ShikOfTheRa, 2016-08-25:
FIXEDWING uses FORCESENSORS which enables them all and overrides MSP_STATUS,
But then BARO data should be overwritten by GPS data

unless..
USEBAROALTITUDE is defined 

It isn't by default and shouldn't be if no BARO installed.

Maybe there is a little bug in there somewhere

jflyper, 2016-08-25:
Ah, I understand.

Just out of rotorwing's curiosity, why FORCESENSORS for FIXEDWING and always override BARO with GPS unless USEBAROALTITUDE?

To me, something like

```
if (BARO is not available) {
    baroalt = gpsalt;
}
```

make more sense.

ShikOfTheRa, 2016-08-25:
I think it was a quick and dirty hack at the time and grew to appease all the various requests....
Up until PatrikE's fixedwing became popular, all MWOSD users running GPS were using BARO/MAG so it was never an issue.

fixedwing uses GPS only for its operation, but it doesn't replace the BARO/MAG values with the GPS values via MSP. 
And then some user complained because not having BARO/MAG and OSD not working.
So FORCESENSORS introduced to accommodate all - and allow those that prefer to see the MAG/OSD info on display. And the positive was it means the default layouts worked. 
Whole bunch of minor issue really. Am sure it could be better. 

Also any changes would have to consider GPSOSD and NAZA and other non MSP_STATUS options.

ShikOfTheRa, 2016-08-25:
in among teh above... it could be implemented better! But it works...

sppnk, 2016-08-25:
Really not a big problem, I would prefer your efforts in implementing Wh/km efficiency calculations (see #112). :-)

jflyper, 2016-08-25:
@ShikOfTheRa Got it!!! I really appreciate your time!!!