ShikOfTheRa/scarab-osd #301: Add Support for RaceFlight

Tomcatt, 2016-08-26:
Are you able to link up with the RaceFlight group so that we can have telemetry on our OSD along with programming PID's and rates?

I like your work on BetaFlight so it would be a great addition to see it also with the RaceFlight group.

tracernz, 2016-08-26:
Have they invented yet another protocol? Otherwise it should already work?

Tomcatt, 2016-08-26:
I'm not a coder so I can't say.  Everyone on slack say there's OSD as of now.  I can find an invite to slack RaceFlight and you can talk to 2 of the dev's I see on there.  If there's more I haven't meet them yet.

dustin, 2016-08-26:
Can this bug include something concrete?  What has raceflight done to make
themselves incompatible with other systems?

Tomcatt, 2016-08-26:
http://raceflight.net/invite

Join us on Slack.  I'm sure you guys could make something killer.

Sent from my 📱6s+

> On Aug 26, 2016, at 23:50, Dustin Sallings <notifications@github.com> wrote:
> 
> Can this bug include something concrete? What has raceflight done to make
> themselves incompatible with other systems?
> 
> 
>

dustin, 2016-08-26:
I don't think a bug is an appropriate place to advertise a private chat.  I
don't think it's beneficial to limit the discussion and take it away from
the participants.

If there's something specific you need done, please provide a design doc or
similar that can be consumed by anyone who might want to work on it any
time.

ShikOfTheRa, 2016-08-28:
Can someone advise what needs to be added for raceflight?

jflyper, 2016-09-01:
Not much info as far as a casual googling told me.

Activities on RF is rather vague; core people seem to be rewriting everything or that's at least what I can tell. Hope they come up with an eye-popping feature request for OSDs.

proggod, 2016-09-01:
Honestly we aren't exactly good at following up on these githubs l, but if someone running rf wants to test out you osd and diagnose the issues hit us up.  Someone said all the packers etc all looked fine but the osd didn't work

jflyper, 2016-09-01:
So, this PR is not a feature request, but a support request???

ShikOfTheRa, 2016-09-01:
Doesn't seem much development of raceflight in last few months. 
Personally I would love to support it but don't want to waste time if its not being developed anymore.

Need someone to test and give some pointers as to what doesn't work

Tomcatt, 2016-09-01:
They are on a new build.  That GitHub version is a ghost town now.  You need to get an invite to chat on slack.

proggod, 2016-09-01:
If venture to say we are most active project at the moment.   Multigp finals this weekend will probably be mostly raceflight

ShikOfTheRa, 2016-09-01:
OK - glad to hear. Must admit I haven't seen anyone running it here yet.

Not fussed on the TOC for signing up so not for me...

proggod, 2016-09-01:
Not sure what your last sentence meant?

ShikOfTheRa, 2016-09-07:
I started to read the terms for using Slack but too much text to see what I'm signing up to so I'll probably give it a miss.

proggod, 2016-09-07:
Slack is a multi-billion dollar company used by most of the internet :)

ShikOfTheRa, 2016-09-07:
Learn something new every day. Never encountered it before  I did watch the demo - looks good group collaboration option if its what needed. Why not use github like everyone else for files though?

Anyway - happy to add in support if someone can tell me whats needed. Better still create a PR!

proggod, 2016-09-07:
slack is a million times better for collaboration and to get things done quickly.  Its also turned into a great community and support for our users.  As for our code, we use an internal gitlab server.  Everything we do isn't for the public, and I just prefer gitlab over github.  Our software is distributed now on our website raceflight.net

ShikOfTheRa, 2016-09-07:
I can imagine. 
I'm in so many different collab tools it p..s me off. My company even sells collab tools. I have to use them...
I hope that market consolidates! Looks great for dev team though ...

Invite only?

proggod, 2016-09-07:
You can invite yourself at raceflight.net/slack  and anyone is welcome, we have a very large community on there.  Scrolls nonstop in main channels.     Everyone seems to be moving to slack lately, skype even launched something trying to compete

Its great because they have clients for all OSes and phones

dustin, 2016-09-07:
Slack is just IRC for people who haven't heard of IRC.  There's nothing
particularly interesting about it and i don't know why you insist on coming
here to advertise it. This is not the place.

It's fine if you want to use different tools than everyone else.  Do what
makes sense to you.  Just don't be surprised when your can't convince
everyone to rush your island when you want something from them.

This is the bug tracker for a software project.  If you are capable of
communicating what you need from this software, this is the place to do
it.    As that does not seem to be the case, I recommend closing this bug.

proggod, 2016-09-07:
Because the project owner asked?   Am I missing something, is @ShikOfTheRa not the owner of the project?

dustin, 2016-09-07:
In which conversation did @ShikOfTheRa ask you to come here and advertise your service?

If you are actually a "prog god" you surely are capable of writing a protocol spec and dropping it off here and having programmers write code for it -- or possibly even contributing some code yourself.  However, repeatedly asking for the conversation to be move elsewhere is just inappropriate and a waste of people's time.

This bug has been open for two weeks without even the slightest hint of what you need done.  It's indicative of a messy, dysfunctional project that can't even state at a high level what its goal is, much less what steps are necessary to meet it.

"Just go to slack" isn't the answer.  It's not the right type of place to do inter-project collaboration.  It's an OK communication tool for having an interactive conversation between two people in the same timezone.  It's not, however, "used by most of the internet" -- current user estimates are a fraction of github, which is a niche used by developers to collaborate on building software.   More people use Google+.  That doesn't make it useful.

This project is on github, so if you need something from this project, this is the right place to ask for it.

12 days ago, @tracernz asked if you've invented a new protocol and that's what support is needed.  If so, could you state that?  Can you point to the protocol specification?

If not, are you able to say what protocol you are using?  If it is supported by MWOSD today and doesn't work, can you say why not?  What else needs to happen?

If you can't provide a spec, or like, goal, can we close this bug now and stop using this space as advertisement?

proggod, 2016-09-07:
Here is raceflights official stance.  If someone wants their product to work with raceflight and are having issues, they need to come to slack and discuss it with our developers.  we have no time to watch 500 different githubs, nor will we.   So I am done with this conversation.   We are by far the fastest growing flight control software, and we LOVE working with people.  But we have way too many things going on, and way too many developers to spend time chasing every issue, without some sort of quick communication.   Its a shame people like you are involved in any of these projects, all you do is chase away the good developers.  The good news is a lot of them end up with us, since we are welcoming to anyone, and would never hope to attack someone just because they are trying to work with someone else to get their software compatible with ours.  This will be my last response to this ticket, and the ticket, so as far as i know scarab-osd isn't working, and anyone wanting to run raceflight with an osd, i suggest you look for other options to which we have better collaboration with the owners.  We are currently working directly with just about every major hobby company on products, its sad that we are getting such a welcoming response here.

tracernz, 2016-09-07:
That's an unfortunate stance. 😿You must understand that Github has been (and still is) the tool of choice for code and issue tracking for the majority of open-source projects, people are happy with it. As for real-time communication, IRC has been the established winner there since longer than I have been alive, slack is just another 5-minute wonder at this stage.

rs2k, 2016-09-07:
dustin and tracernz, I don't know what your problem is. proggod offered to help with the issue Tomcatt created and you two are jumping down his throat over how our team opperates. This attitude is not tolerated by the RaceFlight team. We have a large and tight-nit developing team who choose to be able to communicate in real-time without worrying about the threat of being belittled for a bad idea in public. We wish to encourage top-level development from top-level minds and are more than happy to work with Tomcatt and/or ShikOfTheRa as long as everyone can remain civil and fair.

dustin, 2016-09-07:
@rs2k I don't think anyone offered an opinion about how your team operates.  In contrast, you guys have come here and demanded anyone who might be able to help you change the way we all work.

This is a bug tracker for a project that predates raceflight.  Someone is here asking for people on this project to do free work to support your, presumably, commercial venture.  You aren't publishing source code we can fix up (as I have with other projects to support MWOSD), protocol specs we can comment on, or a concrete bug report saying what's going on.

> We have a large and tight-nit developing team who choose to be able to communicate in real-time

Some of us have jobs and live on different continents.  Many of us work in [real engineering organizations](https://github.com/google) where we're used to seeing things like specifications and design documents before just throwing code at things.  At least to start conversations.

> without worrying about the threat of being belittled for a bad idea in public

We're not belittling you about your bad ideas, we're just asking you what they are so we can know what you want us to do.  There's nothing actionable in this entire bug report.  It's not even clear what "Add Support" means.  Is it a new protocol?  Is it a `#define` that just advertises your name?

> We wish to encourage top-level development from top-level minds...

I hope you can recognize that this isn't particularly encouraging.  @tracernz is absurdly prolific and has been doing some work lately that would be tremendously beneficial to you.  If you look at my github profile you'll see that my open source work is also pretty clearly "top-level."

We are literally, genuinely trying to help you by asking you what you need done and you're pushing us away because (as far as I can tell from reading the above) you don't want to embarrass yourself if your first idea isn't good enough?  It's OK to have bad ideas.  They're pretty much required for having good ones.

But you can't reasonably expect every system that ever interacts with your software to change how they do development just because you want to do things differently.  At some point, you have to extend some olive branch towards interoperating.  "Register to get our commercial software" and then "Register to use our proprietary chat room to do an off-the-cuff design" is not feasible.

This bug has been open for two weeks and there is no signal at all indicating what you might want done.  But I'll go ahead and step out now.  I'm not going to register a copy of your software just to see if I can improve it, so I guess I have no dogs in the fight anymore.

tracernz, 2016-09-07:
Couldn't have said it better than @dustin did. Good day chaps, and good luck with your venture. :smile: @ShikOfTheRa might be willing to spend his time jumping through your hoops to help your project, and if so that's fine, but I hope at least you can understand a little why people have an issue with the way you try to bully other projects into adopting your methods over the already established ones. :wink:

ShikOfTheRa, 2016-09-08:
LOL at you keyboard warriors. Clearly there is some history here. 
I’m going to close this and open a  new one without all this .

PS - Anyone want a dig at me – fuck off. I will not respond or engage… Can’t be arsed. I have a life and I am thankful it is so good compared to what many people in the world have to endure. Not worth wasting any more time.

ShikOfTheRa, 2016-09-08:
#313

will be the new clean version of this thread.

closed 2016-09-08 by ShikOfTheRa