ShikOfTheRa/scarab-osd #302: Support for voltage warnings per profile.

riquezjp, 2016-08-28:
Is it possible to have separate voltage warning based on the current profile?

For example: I have profile 1 for 3S battery & profile 2 for 4S, but I would like to set different voltage warning levels for each.

jflyper, 2016-08-28:
To complement, this feature request has been derived from a discussion on fpvlab thread:
[http://fpvlab.com/forums/showthread.php?34250-MWOSD-for-MULTIWII-NAZE32-BASEFLIGHT-HARIKIRI&p=809483#post809483](url)

**Riquez**:
> Is it possible to have separate voltage warning based on the current profile?
For example: I have profile 1 for 3S battery & profile 2 for 4S, but I would like to set different voltage warning levels for each.

**teralift/jflyper**:
> Unfortunately, no.

We may be able to have several set of cell count and warning levels on OSD and switch around them as profile on FC changes.

However, the cell count and warning levels do not synchronize with what you set with FC configurator. This is because there is no message defined that carries these information from FC to OSD. The profile number is different; it is sent in MSP_STATUS message, along with arming status, available and active sensors and such.

Why don't you submit a feature request in the MWOSD github, in issues page? (Does not guarantee it will be implemented, though, due to lack of time and space.)

ShikOfTheRa, 2016-08-28:
Support for this already exists.
Undefine AUTOCELL and AUTOCELL voltage.
set alarm on gui to 34 for 3.4 volts per cell

It needs tidying up but never seemed much interest so I haven't progressed.

/********************       Voltage Warning Settings         ************************/
#define AUTOCELL
#define CELL_VOLTS_WARN 35        // Specify the cell voltage level at which low voltage warning takes place eg. 35 = 3.5 volts per cell
#define CELL_VOLTS_MIN 34            // Specify the cell voltage at which it is considered empty
#define CELL_VOLTS_MAX 42           // Specify the max normal LIPO cell voltage
#define AUTOCELL_VOLTAGE           // Optionally use Main battery Alarm value as individual cell value. i.e. 3.4 = 10.2v on a 3s

ShikOfTheRa, 2016-09-14:
There is an option to support this.
Operation amended, config.h tidied up and made hopefully clearer.

See following section
/********************       Voltage Warning Settings         ************************/
//#define AUTOCELL                  // Uncomment this to use varying cell count batteries. Overrides GUI/OSD voltage warning settings. Uses CELL_VOLTS_* below unless AUTOCELL_ALARM defined
/

closed 2016-09-14 by ShikOfTheRa