ShikOfTheRa/scarab-osd #303: No overlay until camera is unplugged and plugged back in again

theArchLadder, 2016-08-29:
I know similar issues to this has been posted, but i have not found any issue exactly like this and more importantly, i have not been able to find any solution.

I have this camera:
http://www.foxeer.com/XAT600SM-Super-Mini-Sony-600TVL-Camera-g-15.html

When i power up my quad, i get no overlay, if i then unplug the camera i get the overlay, and i still have it when i plug the camera back in again.

I have tried with:
//#define AUTOCAM 
//#define USE_VSYNC
And tried both PAL and NTSC in the GUI settings and power cycled after setting changes, only PAL works as it is a PAL camera, i also tried:
#define FASTPIXEL
But it made no difference.

It seems like unplugging the camera triggers some kind of calibration/detection that make it work, but what? And how can i get this to work without having to unplug/plug in the camera? Can i hard-code any other values?

Please let me know if there is anything i can try or if i can help with debugging in any way!

dustin, 2016-08-29:
That sounds a lot like my experience in #241

theArchLadder, 2016-08-29:
@dustin Perhaps, but as far as i understand you got some success with the AUTOCAM feature...? I always have the same issue and have to replug the camera, and i still don't have any solution.

Oh and btw, i cloned the github repository so i am running the latest stuff.

dustin, 2016-08-29:
I thought I did, but didn't.  My problem did end up going away, though.  :/

jflyper, 2016-08-29:
Those of you having a problem with CAM detection, can you test the latest dev to see if it solves your problem?

[https://github.com/ShikOfTheRa/scarab-osd/tree/master](url)

jflyper, 2016-09-04:
@theArchLadder 
I tested my HS1177Ms, and they work beautifully with the current dev master.
I advise you to give it a try.

https://github.com/ShikOfTheRa/scarab-osd/tree/master

Please don't forget the usual customizations in Config.h.

ShikOfTheRa, 2016-09-14:
I have also tested latest with numerous cams - PAL / NTSC all work good with exception of the nexchip ones that have incompatibility issue with all MAX OSD.

Don't think can do any more with this from a code perspective so will close.

closed 2016-09-14 by ShikOfTheRa

jnrabreu, 2017-06-15:
Hi!

I quick update on this subject as it probably can be fixed.

I had it working on my osd until i updated to 1.7.1. I went back to 1.6 and made severel tests and found that if enabled: 
#define AUTOCAMWAIT               // **UNTESTED** - Use with AUTOCAM - waits until camera is ready - i.e. if power up cameras after FC. 
I got OSD working again with Foxxer XAT600SM

This option is not available on 1.7. Is it possible to get it back ?

ShikOfTheRa, 2017-06-16:
Yes. Am surprised though -  it shouldn't be needed - the updated code looks for change rather than wait. It effectively does same thing automatically.

jnrabreu, 2017-06-16:
That is great as i have some cameras that don´t work with 1.7.1. Those cheap Bannggod HS1177 clones have the same problem also. With 1.6 and AUTOCAMWAIT they work fine, but not on 1.7.1

ShikOfTheRa, 2017-06-16:
Just had a look and can you try this on 1.7.1 - it should do same thing.. 

Disable AUTOCAM
Set correct NTSC / PAL in GUI

#define AUTOCAM                     // Disable if no screen display. Enables autodetect Camera type PAL/NTSC. Overrides GUI/OSD settings.

jnrabreu, 2017-06-19:
Tried it but  dosent work.

ShikOfTheRa, 2017-06-19:
OK - will re-add the wait. prob wednesday as tied up in work at moment.
This is a strange one - i also have some 1177 clones.