ShikOfTheRa/scarab-osd #304: Soft resetting MAX

jflyper, 2016-08-29:
Any one have experience with successful software reset of MAX?

It's harder than I thought it would be.

Actually, it's not the software reset facility, but how to bring the MAX out of any intermediate state that is really hard.

tracernz, 2016-08-29:
MultiOSD: https://github.com/UncleRus/MultiOSD/blob/master/firmware/lib/max7456/max7456.cpp#L177

I recently wrote a driver myself (for us on PlayUAVOSD) and just followed the datasheet re: time to wait after writing the byte, before polling the status register until the reset is complete. Didn't encounter any issues?

jflyper, 2016-08-30:
@tracernz Yeah, that's the easier part of the job.

Problem I'm trying to solve is:
- M328 has RESET line connected to DTR, so it can be reset at arbitral point in a program.
- MAX don't have RESET line connected.
- Power on reset, M328 and MAX successfully runs the program.
- Later, M328 is reset by pulling RESET low.
- MAX don't reset this time.
- M328 tries to use the software reset bit via SPI, but realizes it does not remember in what stage of the SPI transaction it was at the time of the reset. It may be any one of the following:
o SPI was idle.
o Just written first byte (register and r/w designation) on to the SPI.
o Waiting for next data byte in auto increment mode.
- Challenge is to bring the MAX to idle in terms of SPI transaction so I can use the SPI transaction to set the software reset bit and observe various status.

tracernz, 2016-08-30:
I understand now. :+1: 

From the datasheet:
>  If CS goes high in the middle of a transmission, the sequence is __aborted__ (i.e., data does not get written into the registers).

jflyper, 2016-08-30:
>From the datasheet:
>>If CS goes high in the middle of a transmission, the sequence is aborted (i.e., data does not get written into the registers).

Oh, hey, this is a great information. I was just wondering if I have to write a bit-banger on SCK line that pumps out incomplete byte out of the slave shift register!!!

I can concentrate on the transaction itself now!

jflyper, 2016-08-30:
Crap!!! M328 pins goes low on reset (actually floating input, but looks like low), so it doesn't terminate the sequence.

jflyper, 2016-08-30:
Okay, I think I have a working code now.

Hundreds of lines of test code and several thousands of resets turned into just three lines of code :eek:

```
  // The magic triplet
  spi_transfer(0xFF);
  spi_transfer(0xFF);
  spi_transfer(0xFF);
```
Actually, there were two cases: one `spi_transfer` or three `spi_transfer`s. For one transfer case, additional 'read from register `0xff`' seem to do no harm, hence three transfers.

The value `0xFF` probably is essential for the first transfer; it terminates the possible auto increment mode.

Strange thing is, that this triplet doesn't work if inserted at arbitral point in the execution; it only works when I really want to reset the MAX :confused:

There's no concrete theoretical explanation why it works; more experiments on behaviors of the MAX under invalid register designation, incomplete transaction, etc. is required, but kinda tired for now. May be sometime later ;->

closed 2016-09-01 by jflyper