ShikOfTheRa/scarab-osd #305: Runtime error feedback (feature proposal)

jflyper, 2016-08-31:
What do you think about optional but explicit feedback of runtime errors?

Possible feedback channels are:

1. Text displayed as OSD overlay.
This is already implemented for some errors, such as "NO DATA".

2. LED based error signaling

- The current software defines an LED on D7, which is used for heart beat. It's too bad that not many, if not none, commercially available hardwares have the D7 LED.

- Alternative would be the one on D13, which is shared with SPI SCK.
Using D13 explicitly to drive an LED prevent access to MAX, so screen update will be skipped while the signaling

3. Send errors to GUI
It only works when connected to GUI, but still work for on-the-bench. Actually, some operational status are already sent to GUI and displayed (e.g. sensor values).

ShikOfTheRa, 2016-09-01:
Yep - makes sense. Would prefer if can be enabled via #define - anything specific thinking of?

D7 interesting - it wasn't on the first cut of schematic for my variant either It is now!

dustin, 2016-09-01:
Does it make sense to unify that with the msp alarm extension I added?
This seems similar, except the flight controller is telling you something
is wrong, and how wrong it is.  At least from a UI perspective, having one
thing displaying messages from any of what is one of three or four things
would be beneficial to the user.

This could include:

* No data
* Low voltage, etc...
* msp originated alarms (e.g., dRonin telling you your RX is in failsafe)

These each have severity levels which can help drive LEDs as well, though
it may be useful to have a sort of source indicator there.

jflyper, 2016-09-01:
@dustin I was thinking of OSD specific runtime errors that help people trouble shoot problems associated with OSD, but yes, some of the implementation possibilities naturally implies your idea. I will give it a thought and try to organize some rough design.

jflyper, 2016-09-04:
@ShikOfTheRa 
The D7 LED has been in the code (LEDPIN) for as long as I know:)
Currently, it is only providing 1Hz heartbeat in the main loop, and another heartbeat during font uploading (probably there to keep it flashing).

ShikOfTheRa, 2016-10-06:
No sure if any interest here but as an FYI, there were some changes to accommodate alarms messages better:

If the message is fixed and not something coming in via serial, can use alarms active bit method
so setting  alarms.active|=(1<<6); would display low volts as per table as below until its reset. 
To avoid overwriting, it display error messages one at a time and loops through them whilst alarm bits are set.
 
const PROGMEM char * const alarm_text[] =
{   
  blank_text,     //0
  disarmed_text,  //1
  armed_text,     //2
  nodata_text,
  nogps_text,
  satlow_text,
  lowvolts_text,
};

ShikOfTheRa, 2017-01-10:
Been no activity here for a while so will close.

As above there are some alarms.
Have also added a debug screen and text message option. Please see top of def.h for options

closed 2017-01-10 by ShikOfTheRa