ShikOfTheRa/scarab-osd #306: MW_OSD for MacOS

Tomcatt, 2016-08-31:
I downloaded the updated Github file from here to update my Micro MinimOSD.  The Mac version isn't working.  Do I need to compile it?

jflyper, 2016-08-31:
Can you revert to the R1.6 GUI? It should be compatible with R1.6.1.0 and latest dev master.

Rodrigoah, 2016-08-31:
Same here, will try R1.6

jflyper, 2016-08-31:
@Tomcatt 
> MW_OSD_GUI is damaged and cannot be opened...
Do you have this problem, too?
If so, you can fix it by modifying the permission bits.
See FAQ#4 in DOCUMENTATION for details.

Tomcatt, 2016-08-31:
Yes, I still have the older copy.

I'll checkout that doc file tomorrow and let you know my findings.

Rodrigoah, 2016-08-31:
@jflyper Solved with FAQ#4... Feel soooo dumb...

ShikOfTheRa, 2016-09-14:
@Tomcatt are you OK now?

closed 2016-10-20 by ShikOfTheRa

Tomcatt, 2016-10-20:
Okay, I'll test it out later today.