ShikOfTheRa/scarab-osd #308: Working but not working

Tesla63, 2016-09-02:
Lux FC mico minimosd, I am getting partial telemetry such as Bat voltage Rssi and flight time, however prior to arming I can see my flight mode and it says disarmed, upon arming it still says disarmed and no data, my flight modes vanish however I still have battery voltage and rssi. I have checked the rx/tx and they are correct as if I swap them I get no data at all. On the lux I am using uart 1 as the manual suggests msp is on and telemetry is set to Frsky (I am using a D4R-II) I have tried various speeds on the ports but no luck. Any help would be appreciated.

jflyper, 2016-09-02:
So to summarize your configuration, you have UART1 configured as MSP and FrSky telemetry.

If this is the case, as soon as your Lux is armed, MSP is shutdown and telemetry will start and feed FrSky telemetry.

Looking at this from the MWOSD, it can communicate with the Lux via MSP while disarmed, and as soon as it arms, the communication will no longer possible, thus NO DATA (perfectly explains your case).

Tesla63, 2016-09-02:
That is awesome however when I turn MSP off I get nothing when armed or disarmed. We know that the telemetry is there because we can see it if MSP is enabled. I am obviously doing something wrong with the Lux as I have had a ton of these all working on the Naze.

jflyper, 2016-09-02:
You will see NO DATA all the time if you disable MSP.

MWOSD does not understand FrSky telemetry. You will see NO DATA without working MSP.

Tesla63, 2016-09-02:
Solved my issue, set UART 2 to MSP on and Telemetry to disabled. works perfect now , thanks for all your help.

closed 2016-09-02 by Tesla63