ShikOfTheRa/scarab-osd #309: incorrect character placement

johancilliers, 2016-09-04:
hi all 
I have a slit problem with the character on my display, I am use micro minimosd with naze32 custom broad, runcam sky2 camera and fatshark 250mw TX hardware and mwosd R6 software. I have the hardware and software install to the letter. 
So it all works fine but some font character are not correct places i.e. the main voltage battery symbol is a not battery symbol but letter and the V for voltage is a - dash . Some other symbols are incorrect too.

How can i fix this issue?

thanks

jflyper, 2016-09-04:
Can you re-upload your font using the MWOSD_GUI?

Notice that you have to explicitly press UPLOAD button after selecting your font.

A window pops up to tell you that the font you have selected is loaded, but this message should not be confused with uploading to OSD is done.

johancilliers, 2016-09-04:
Yes, I have uploaded the font using the UPLOAD button, it showed writing to the broad 256/256 font, I have also cleared the board then re-uploaded using a different font same issue but different symbols appeared, on all font.

jflyper, 2016-09-04:
Okay ... Try uploading a sketch and font from freshly downloaded release package.

If the problem persist after this procedure, your micro has either a bad flash, or bad character memory.

ShikOfTheRa, 2016-09-06:
Also check the FAQ - tehre are a few tips in there. Especially voltage realated

closed 2016-09-14 by ShikOfTheRa