ShikOfTheRa/scarab-osd #310: MAX SPEED is vulnerable to GPS glitches?

jflyper, 2016-09-04:
Originally reported in fpvlab:
[http://fpvlab.com/forums/showthread.php?34250-MWOSD-for-MULTIWII-NAZE32-BASEFLIGHT-HARIKIRI&p=810319&viewfull=1#post810319](url)

> **Schalonsus**
Hm i think something went wrong with max speed. I cant remember flying 309km/h. It was 40km/h wind but 309km/h with skyfun is unrealistic. Maybe it has something to do with bungee start? Or gps glitch? 

[image]

> **teralift/jflyper**
I took a look at the code; it's a simple max substitution from the gps speed. Some glitch is most likely, but 309km/h is 85m/s ... rather big glitch in 3d fixed outdoor condition. How do you evaluate the bungee start? Is it strong enough to maintain average speed of 85m/s for 2 seconds?
 
 
> **Schalonsus**
My bungee is not that strong. I dont believe it does 85m/s.
I always start the skyfun with bungee and this would have been the first time out of 10-15 starts this error occurs. 
I will have a look on that but maybe this was just a one time error. 
 
 
> **teralift/jflyper**
If it happens once, then it will happen again. The current max speed code is too vulnerable to gps glitches; it just replaces the max value if a new value is greater than the current max. May be we should have few more lines of code to detect and discard sudden change in speed. Acceleration based or something. I'll submit it as an issue.

jflyper, 2016-09-05:
I first thought that while non-nav-oriented bleed would feed raw GPS data, nav-oriented bleed such as iNav would feed some cooked (smoothed, filtered) data, so did some code digging.

1. `MW_OSD.ino` is requesting `MSP_RAW_GPS`, and that's the source of the ground speed.
2. In iNav, MSP_RAW_GPS is serviced through `gpsSol` (`gpsSolution_t`), which is set directly by gps NMEA/UBX parser.
3. In iNav, estimated position is stored in `posEstimator.gps` (`navPositionEstimatorGPS_t`)

It's relatively easy to modify the iNav code to use the `posEstimator.gps` instead of `gpsSol` to serve `MSP_RAW_GPS`, I'm not sure if this would break the semantics of the GPS data displayed on OSD.

ShikOfTheRa, 2016-09-06:
Put a temporary simple filter. With low code overhead. Put in a max increase value.
I don't want to filter unless have to because of those record breakers whose 1 kmh / mph makes all the difference!
See how this works... 

I see this issue quite a lot in testing - usually at startup and very rarely in action (I assume some sat glitch).
Normally end users do not see it because it is while they are waiting for good sat count and FC is disarmed. When disarmed, it sets speed to 0 (and hence max speed). 
It was put in for that very reason!! If you take it off you sometimes see the speed vary significantly. And very often upto 3 or 4mph just sat static.

jflyper, 2016-09-09:
How about testing for invalid acceleration (m/s^2) or invalid jerk (m/2^3)?

```
long acc;
acc = (newSpeed - lastValidSpeed) / (lastValidSpeedTime - millis());  // m/s^2 * 1000
if (acc > 9800 * 3 || acc < -9800 * 3) {
  // Invalid acceleration
} else {
  // Valid acceleration, valid newSpeed
  lastValidSpeed = newSpeed;
  lastValidSpeedTime = micros();
}
```

DzikuVx, 2016-09-09:
About GPS glitch detection, you might want to take a look here iNavFlight/inav#136 and iNavFlight/inav#512

ShikOfTheRa, 2016-09-09:
Yes real fix is at FC level.

How to implement a secondary check with minimal memory overhead is the question for us on the atmega. I'm prepared to accept less than optimal to conserve memory.

What I see seems to be crazy big values rather than a few km/h different. 
So I would have thought we should see a max differential of 30kmh in a 1hz update GPS. That's some serious acceleration equivalent to fastest cars on the planet.

Another simple option would be to if "change">"max speed change", limit max speed to previous "max speed" + "max speed change" . We could drop max change value significantly. Carry out some testing to see what fastest acceleration I can get in real world!

I was thinking that in those scenar

ShikOfTheRa, 2016-10-20:
So we have something added to minimise impact. I'm Ok with that considering memeory constraints.

closing :)

closed 2016-10-20 by ShikOfTheRa