ShikOfTheRa/scarab-osd #311: Standalone OSD Vario

JocB1, 2016-09-06:
I used the osd with cleanflight. Now I changed to Graupner GR-18 and I can only use the osd as standalone type for voltage, timer and throttle positon. 

It would be very nice if a Barometer (BMP085, BMP180, MS5611) could be connected to I2C for altimeter, or alternatively a xxDOF Module for altimeter, artifical horizon and compass). 

Is something planned to be added in the future? 

Jochen

ShikOfTheRa, 2016-09-06:
Always wanted to but for me its unlikely  due to time.
Could always connect it up to smallest micro FC you can find... not ideal, but an option.

@jflyper 
Option feasible with your planned code?

I was thinking of accelerometer for the horizon / pitch / roll info, but

jflyper, 2016-09-06:
@ShikOfTheRa 
Feasible, yes! My code is only used in slave mode connecting with FC.

Since this is the standalone mode, our brand new shining I2C facility can be used to connect as many slave devices you like, ACC, GYRO, BARO, PITOT, TEMP, REV, ...

ShikOfTheRa, 2016-09-06:
I'd love to and its right up my street, but know I'll never get time to do it. 
Up to you if have time...

ShikOfTheRa, 2016-10-20:
This has been added to the long term development wishlist / roadmap.

No immediate work or update planned so closing this for now.

closed 2016-10-20 by ShikOfTheRa