ShikOfTheRa/scarab-osd #314: Standalone OSD GPS add HDOP

Hyprmtr, 2016-09-09:
Hello,

Forgive me if this is not the proper place to make this request. Would it be possible to add HDOP to verify GPS connection quality?  This could possibly help mounting locations of the GPS antenna itself and/or identify a sub par GPS amtenna/receiver. 

Thank you,
Ken

jflyper, 2016-09-09:
@Hyprmtr 
You've landed in a right place!

Hyprmtr, 2016-09-09:
Great! Thank you.

ShikOfTheRa, 2016-09-15:
@Hyprmtr 
I have added some functionality for this - please see latest update in the repository

Minimum DOP required for GPS sat fix - For UBLOX only...
Requires minimum pdop value to achieve a satfix
It forces satcount to be 1 less than required minimum. Will probably
change this to HOMESATFIX

Until confirmed working:
1 pdop can be displayed by enabling temperature in the layouts setting.
2 if enabled as per 1 , pdop will display if higher than minimum
required.
3 current default is set to 300 which equals 3.00
4 enabled by default in config.h will move to def post testing.

Hyprmtr, 2016-09-15:
Great! Thank you Shikra. I will give it a go this evening on the bench and fly it in a couple of days.

ShikOfTheRa, 2016-10-06:
Any chance to try out? Have had some positive feedback about HOMESATFIX

Hyprmtr, 2016-10-06:
Hello,

I did attempt to download and flash the firmware with the added DOP but, I couldn't find where to define DOP in config h?  Maybe I didn't find the correct firmware that you changed. I didn't want to post a bunch of daft questions so I just moved on to another project.  I would like to give it a go as my other projects are now complete and I do have time currently. 

Ken

ShikOfTheRa, 2016-10-06:
No worries.

Step 1: In the very latest, enable these:
//#define GPSDOP 300                // UBLOX only - minimum PDOP required for sat fix. A DOP of 2.00 is considered good 
//#define DISPLAYDOP                // UBLOX only - display DOP when its bad - above GPSDOP. Uses and requires temperature position to be enabled

Step2 : in layout editor, set temperature to enabled for the layouts you are using.

Hyprmtr, 2016-10-06:
That's the line I was looking for in confg h. I couldn't find it. I suppose I didn't have the very latest. The rest, as far as the layout editor, was not a problem at all. 
   I will work on it tonight as I need to dig out my OSD from my little sailplane anyway for some changes. Thank you for following up and all your help.

Hyprmtr, 2016-10-07:
I must not be downloading the correct latest firmware. The only page that I found is the latest downloads page. From there, I only get the same firmware I have already that does not have the HDOP addition. Please forgive my ignorance as I am, by no means, a regular here of github just yet.
I just can't locate the page or link with the very latest download that has the DOP added. 

Ken

ShikOfTheRa, 2016-10-07:
Just go here ...
https://github.com/ShikOfTheRa/scarab-osd

and you will see a green button saying "download or clone" 
That takes the very latest from the repository. 
The ones in the "releases" area are a snapshot at that time. 

I remember the same with github. its only easy and obvious when you know!!

Hyprmtr, 2016-10-07:
Ah ok. I believe I did do that but just after I downloaded it the wife called me in the house for dinner. I will get back on it this evening.  I suppose I wasn't as lost as I thought I was. Haha. Thank you!

Hyprmtr, 2016-10-07:
Ok, I got it working. With 4 satellites in the garage DOP was 3.4+. Hopefully I can fly tomorrow and aquire more sats outside but there is a front coming through so it might be too windy. Thanks for the help!

ShikOfTheRa, 2016-10-08:
Sounds positive. In garage 3.4 sounds reasonable. In truth I am not sure what would be a good setting. I went for a gut feel default of 3.00 and I think my DOSD is 1.6 or 2.0 as default. I think it doesn't need to be super accurate for home fix, but you want it fairly stable.

ShikOfTheRa, 2016-10-20:
OK. Think we are done with this now following feedback. Closing

closed 2016-10-20 by ShikOfTheRa