ShikOfTheRa/scarab-osd #316: Changing Layout->HUD loses OSD HIGH/MID/LOW setting

JoeB99, 2016-09-19:
In the layout editor, select your HUD, enable RC switch, any channel.
Set Low to 3, Mid to 4, High to 5.

Now go and edit HUD 5 in the layout editor, as soon as you change the current HUD, the Low/Mid/high setting is lost.

jflyper, 2016-09-21:
Yeah, I noticed it right after I started to play with MWOSD, but didn't bite me strong enough to have a look into the GUI code. I myself treat the post-editing low,mid,high is like a ritual. Do you think it's severe?

JoeB99, 2016-09-22:
I'm setting up 3 screens, this means that i have to test one at a time which is a pain. It should probably be fixed at some point. The RC channel number should also start at '1' not '0'.

jflyper, 2016-09-22:
@ShikOfTheRa 
Can you add an entry to the BUG TRACKER list?

@JoeB99 
RC channel origin problem has been raised in the past, and should have been fixed by now. (Although I haven't tested it yet.)

ShikOfTheRa, 2016-09-22:
Will do.. am away for a couple of weeeks.
@JoeB99 that is by design purely from my laziness! Quick and dirty job to get it working. I th k easy to change so will look at when get back
As FYI there is a revised gui due out shortly that has this issue resolved.  Ugh clearer and easier to use.

ShikOfTheRa, 2016-12-23:
OK this is done. PC and linux GUI updated. I don't have a mac so not going to be one for a while.

closed 2016-12-23 by ShikOfTheRa