ShikOfTheRa/scarab-osd #317: hubosd firmware request

UncleMcNutz, 2016-09-20:
request for hubosd support, i would like to use mwosd on my hubosd if at all possible as the current gui overlay for the stosd8 firmware is very limited.

jflyper, 2016-09-21:
@UncleMcNutz 
HUBOSD and various derivatives are really nice. I actually was looking at one of them tonight on BG.

However, supporting HUBOSD in MWOSD is very difficult, if not impossible.

Main reason is that HUBOSD employs STM8, which is a completely different from ATmega328 that MWOSD is currently supporting. It's like porting a software running on a Mac to a Windows PC.

Slightest possibility is that if there is an Arduino compatible IDE for STM8, then we might be able to do it. I have done the similar porting onto Teensy 3.1 using Teensyduino. It actually was an overnight job thanks to the rich resource provided by 32-bit ARM core on Teensy 3.1, but porting to STM8 would be more painful due to the 8-bit core, and it's no trivial task.

UncleMcNutz, 2016-09-22:
thought it may have been difficult, though regarding pdb's and osd's on one board do you know of any that supports mwosd? i've looked at this [one](http://www.banggood.com/MXK-Power-Distribution-Board-PDB-with-OSD-BEC-Output-5V-12V-3A-LC-Filter-for-CC3C-Naze32-F3-p-1031375.html?p=LI16170938632012117U) but it's bad, has bad reviews and is the only one I've found so far with mwosd compatible components.

UncleMcNutz, 2016-09-23:
https://flyinglemon.eu/glowna/25-lemoncore-pro-pdb.html have a look at this it's an atmelmeag32p pased chip so it should be able to be flashed with mwosd? though i'm not too keen on the price for a chip that size it's seams overly expensive.

jflyper, 2016-09-23:
@UncleMcNutz 
Yeah, I've took a look at it, but it lacks MAX7456 OSD chip. I think it's using LM1881 video sync generator and M328 is explicitly injecting overlay video signals.

UncleMcNutz, 2016-09-23:
damn all these limitations for the perfect pdb osd mix

jflyper, 2016-09-23:
Yeah, the current trend is the FC+OSD integration.

I suppose you have an FC that you really want to stick with?

UncleMcNutz, 2016-09-25:
im flying an xracer f303 v3.1 love it to bits, i don't really like having a lot of stuff on the fc as it always seems to add cpu latency, and with the weighted pid code of betaflight 3.0x you need to trim the fat a bit so it doesn't overload the fc

jflyper, 2016-10-02:
@UncleMcNutz 
I ran across this

[Integrated Power Distribution Board PDB Built-in OSD Dual BEC for Naze32 F3 CC3D](http://www.banggood.com/Integrated-Power-Distribution-Board-PDB-Built-in-OSD-Dual-BEC-for-Naze32-F3-CC3D-p-1069504.html)

ShikOfTheRa, 2016-10-03:
It also says pre-loaded with MWOSD 1.5 - so should be good

jflyper, 2016-10-03:
Ah, and someone commented "First try, not bad but for a newer version of the PDB there is a lot room for enhancements".

Are you going to give it a try?

ShikOfTheRa, 2016-10-03:
Not for me - I have joined many others and am fed up with JST1.0 connectors. I just read his review - guess he's the same too.
Now prefer direct solder or pcb pins.

jflyper, 2016-10-03:
Ah, I was addressing @UncleMcNutz

ShikOfTheRa, 2016-10-03:
Ah sorry!

UncleMcNutz, 2016-10-05:
i think i did have a look at this before but it's almost as much as my flight controller lol, and i'm really not a fan of jst 1 connectors, was looking at the rrosd which isn't much difference between the hubosd, though flying the hubosd a fair bit more and ive just got used to the minimal style of it, though still would like mwosd for the added functionality of the artificial horizon

jflyper, 2016-10-17:
How about designing your own pdb board with provision to accept regulator and osd components?

[Something like this.](http://sputnikfpv.com/product/micro-pdb/)

ShikOfTheRa, 2016-10-20:
Unfortunately not much we can do her e- unable to support the hardware so closing this as an active request.

:(

closed 2016-10-20 by ShikOfTheRa