ShikOfTheRa/scarab-osd #318: The sight in the center of the screen

a6j, 2016-09-26:
Is it possible to implement a shutdown function Horizon Bar and turn a sight (of the cross) in the center?

jflyper, 2016-09-27:
Can you elaborate?

Do you want to turn AHI (Artificial Horizon Indicator) and Airplane marker OFF? Or switch between ON and OFF?

a6j, 2016-09-27:
Now if you need a cross in the center, you need to enable Horizon Bar. But he did not need a racing copter, but useful sight in the middle.
[image]
[image]

jflyper, 2016-09-27:
You can do it by
(1) Enable FORCECROSSHAIR in Config.h (remove // at the beginning of the line).
(2) Disable Display Horizon Bar from GUI.

a6j, 2016-09-27:
Thank you. Everything worked as needed. But it is much easier to bear this option in the GUI configurator.

jflyper, 2016-09-27:
Glad it worked as you wanted.

> But it is much easier to bear this option in the GUI configurator.

Consider GUI as GCD. Not every option can have a place in the limited screen. (And it's kinda pain for us to keep adding GUI support for ever increasing features.)

a6j, 2016-09-27:
Good. Thank you.

closed 2016-09-27 by a6j