ShikOfTheRa/scarab-osd #320: Standalone OSD, add compass

zingaros, 2016-10-04:
Hello,

I be been testing the 1.6.2 release for a couple of flight as a standalone osd. Has worked flawlessly: well done! I like especially the better accuracy of the home position.

I was wondering if you could add the support of the compass display that is often embedded with the gps module.

Again well done, this osd rocks.

ShikOfTheRa, 2016-10-04:
Possibly in the future. 
The issue we have is hardware support - none of the OSD's I have seen have the necessary connectivity - an I2C and at the right voltage to support. Possible for some with some really good fine soldering skills. 
There may be one or two OSD's that support this in future. At that time possibly can look at that as a fully supported option.
Are you using with fixedwing or a multirotor? For fixedwing, GPS is generally better for home direction heading. unless windspeed greater than airspeed :)

zingaros, 2016-10-04:
Thanks, understood, hardware limitation. 
I will try the fixwing option.
Do you want me to close this?
Have a great day.

ShikOfTheRa, 2016-10-05:
I probably wasn't very clear in my answer - no need to use fixedwing in the OSD config. Actually it probably wouldn't work for you.

What I really mean to say is that with fixedwing aircraft, the GPS reports the direction the plane is travelling and a mag reports the direction the plane is pointing. In wind they are probably not the same... GPS is more accurate to report the home arrow  etc..  The only time mag is better is in strong wind where the windspeed is greater than aircraft speed.

ShikOfTheRa, 2016-10-20:
I have added  onboard I2C compass/baro/accelerometer support to the longer term development wishlist / roadmap.

Nothing planned in short term or furtehr updates here so will close this

closed 2016-10-20 by ShikOfTheRa