ShikOfTheRa/scarab-osd #321: Support for Furious FPV Piggy OSD

Tronnic, 2016-10-08:
Hi, a few weeks ago Furious FPV released OSD Hardware called PiggyOSD. It fits perfectly on the back of most popular cams and is really nice since it has extra ports to configure it without desoldering your RX/TX wires from the FC.

They supply you with an preinstalled version for Betaflight and offer an version for Kiss. Unfortunately the Kiss version is super limited and only offers Voltage and the default Timer. 

It cannot show anything else like Amps, Arming, Modes, Timer (that is connected with Arming) and so on. 

Is there any way to make the most recent MWOSD Beta Version compatible with the PiggyOSD? That would be great!

Here's the Hardware: http://furiousfpv.com/product_info.php?products_id=145
Here is the downloadable Firmware for the Kiss OSD: http://furiousfpv.com/document/kisspigg.rar

Inside the Kiss Firmware Folder are other files. There's also an "Diagram" which probably shows the pin layout for the board (or whatever its called, sorry no native english here). It's located at ..\OSD_Pigy\minimosd-extra-master\board\diagram.jpg

Thanks!

Best regards,
Tronnic.

ShikOfTheRa, 2016-10-08:
No idea. It definitely can...

ShikOfTheRa, 2016-10-08:
I'll check with them....

discoltk, 2016-10-09:
> info.php?products_id=145
> document/kisspigg.rar

PiggyOSD runs MWOSD.    When you install MWOSD (on anything) you need to
configure the type of flight controller software  in Config.h.   (#define
KISS).     The kisspigg.rar is surely just pre-configured that way.    You
probably only need to hook up the MWOSD_GUI and reconfigure it.

Tronnic, 2016-10-09:
discoltk, I of course did that. Please read what I wrote above. Only voltage and timer is working. Nothing else. I wanted to fly with volt, timer, arming, mah, amps, callsign and rssi. It just doesn't work. I did try it out and wasted 5 hours of trying what function works and what does not.

Tronnic, 2016-10-09:
Also there is no way to edit their config since it's an hex file. At least not a way I know of.

Tronnic, 2016-10-10:
[image]

ShikOfTheRa, 2016-10-11:
OK - mailed them too. Happy to work with the guys.

jflyper, 2016-10-15:
> Here is the downloadable Firmware for the Kiss OSD: http://furiousfpv.com/document/kisspigg.rar
Inside the Kiss Firmware Folder are other files. There's also an "Diagram" which probably shows the pin layout for the board (or whatever its called, sorry no native english here). It's located at ..\OSD_Pigy\minimosd-extra-master\board\diagram.jpg

- It looks to me that most of the files in the RAR is not related with MW_OSD.
- diagram.jpg is just a JPG version of MinimOSD-v11.sch.

jflyper, 2016-11-17:
Any progress?

There are several reports on how peculiar the Piggy is, compared to regular Minims.

Hardware problem
A user has reported that a capacitor called C8 should be removed to make his Piggy programmable.

Boot loader problem
Some claim that a special loader that comes with the above mentioned kisspigg.rar must be used to load the manufacturer supplied MWOSD_ino.hex, while the user mentioned in the above "Hardware problem" seems to have no trouble using the regular Arduino IDE to upload MWOSD sketch (a side from it's working properly or not).

flounderscore, 2016-11-21:
I could get MWOSD to work on the PiggyOSD. First, uploading the image seemed to take longer than usual. Then I also had to use the RX/TX pins on the side meant for the FTDI cable to connect to the FC. 

I haven't head time to check whether those pins are connected to the dedicated second set of rx/tx pads.

jflyper, 2016-11-21:
@flounderscore I thought those two rows of signals are both connected to the same rx/tx. If not, the MCU on the Piggy might be M328PB. Can you read the marking on the M328?

flounderscore, 2016-11-21:
I found my problem, "RX" on the piggy-specific pads is connected what's labeled TX on the micro minim osd (the pin next to DTR). 

It seems (not tested, just from measuring connectivity on the pcb) one has to connect FC's RX to RX on the piggyosd.

The Atmel is a M328P.

jflyper, 2016-11-22:
@flounderscore Yikes! I forgot to tell that. There was another one who found it out the hard way, just like you did.

closed 2017-01-22 by ShikOfTheRa

wt29, 2017-02-05:
This is true - the Furious FPV site is wrong. My SP F3 Deluxe needs Rx on FC to RX on Piggy otherwise "No Data / Disarmed" is all I see

ShikOfTheRa, 2017-02-06:
Great huh!! Mistakes happen though.

HenryIvarsson, 2017-02-22:
[image]
FYI

HenryIvarsson, 2017-02-22:
I was curios if this is MWosd, I asked about the source and he could not provide it

jflyper, 2017-02-23:
IIRC, it's full MWOSD. May have some defaults moved.

flounderscore, 2017-03-12:
I'd like to correct my previous statement: Arduino thought it had programmed the PiggyOSD successfully (after an awful long time), but it required me to remove C8 to really get it to work.