ShikOfTheRa/scarab-osd #322: Confused about s.port

discoltk, 2016-10-10:
I'd like to get vario (and possibly other) data from smartport into the OSD.    I'm very confused about the s.port documentation in this repo.

Is it correct that smartport works as a target -> sensor polling and that another smartport device (flight controller) cannot read this data, only provide it to the receiver?     

Any clarification on how to get vario or other s.port telemetry into MWOSD would be much appreciated.

ShikOfTheRa, 2016-10-10:
FRSKY s.port support is currently only in there if you are using an older Multiwii 8 bit flight controller that has patch applied for S.port
I think now no-one is using this. And maybe documentation etc. should be removed to avoid confusion.

I don't use myself and have no plans to implement, but would welcome development for anyone who does.

discoltk, 2016-10-10:
Yea, definitely confusing to have the doc in there.  Maybe at least mark it as deprecated.

I'll have a peak at the code myself, but would it be very hard to pass telemetry values such as altitude as an aux channel, similar to RSSI?

ShikOfTheRa, 2016-10-10:
Well actually it still works - its just that i think no-one uses the old mw controllers and sprt and osd combo this supports... 

FOr sure, need to make it clearer that is supports this old stuff and not new...

With an inverter, its definitely very achievable to add support. Just need someone who can code and has the gear.

There are OSD firmware out there that support FRSKY on the minimosd - maybe one of those would suit?

ShikOfTheRa, 2016-10-20:
Have added this to the development wishlist.

There is nothing immediately planned or to add so will close this as its transferred to the list.

If anyone requires some assistance / guidelines to implement, please contact...

closed 2016-10-20 by ShikOfTheRa