ShikOfTheRa/scarab-osd #323: Where is the code for controlling a 6705 5.8ghz video transmitter?

sonicxx, 2016-10-17:
So, innova and impulse RC both have firmware for this, but  i'm not sure where to find it.  I'm a neofight, where is the best place to look for it Or request it?

ShikOfTheRa, 2016-10-19:
It is a modified version - you may request the source from the store that sells them. 

That source is not available here at the moment.

ShikOfTheRa, 2016-10-20:
@sonicxx 

You may request the source code from the vendor. They are legally obligated to supply it to you if requested as it is based upon MWOSD

Not much else can add so will close this.

Feel free to update of progress. always inteerested!

closed 2016-10-20 by ShikOfTheRa

sonicxx, 2016-10-22:
https://github.com/PickwickGroup/scarab-osd

ImpulseRC

ShikOfTheRa, 2016-10-22:
ah cool. Wonder if they would like to include in core code. Might drop them a note.

sonicxx, 2016-10-22:
If we think about it for even a half a second, bulky dipswitches or a MCU just to light LED and switch channels was behind the times even when it was conceived. 

Makes much more sense to intergrate the 2 functions OSD and VTX.  Cleaner electrically, better spacewise and no physical access needed to the VTX. 

http://impulserc.com/helix-video-transmitter-board

https://www.youtube.com/watch?v=FvCB403GPGY 

25-600mw switching via the OSD and 6s in. Only $40, so why would anyone want the TBS 5.8g where your video takes a round trip...and you don't have OSD.

ShikOfTheRa, 2016-10-24:
Its a good looking board if its what you want and fits your quad. Be interesting to see if it catches on.