ShikOfTheRa/scarab-osd #324: Wrong altitude showed when using LTM telemetry

sppnk, 2016-10-19:
When using LTM telemetry the GPS altitude value is multiplied x10.

ShikOfTheRa, 2016-10-19:
Had a veryquick look through and cant see any obvious issues. 
@digitalentity
Can you confirm if LTM sends altitude in cm?

sppnk, 2016-10-19:
Hi, after some more tests, I have seen sometimes how the value rises. As I am testing LTM throught downlink telemetry at the same time I can see the value in EZGUI software and I see there a normal value.

ShikOfTheRa, 2016-10-19:
Ah did you change to 100? I'll check again.

ShikOfTheRa, 2016-10-20:
Ah yes I see it - It was *10 that confused me.
That helped. Run through again and can see the *100 error straight away :)

Corrected and in latest update in repository

sppnk, 2016-10-20:
Great !! Thank you.

closed 2016-11-25 by ShikOfTheRa

tednv, 2017-01-10:
still seeing this in 1660 what was the change to fix it?