ShikOfTheRa/scarab-osd #326: Cursor and stick input inconsistent

HaseebNain, 2016-10-26:
The cursor seems to be moving inconsistently. It will work as intended, moving one step per stick input but randomly begin to skip selections and jump. 

It looks like it holds the previous input and processes it multiple times.

I have disabled the "hold" feature for testing to ensure that it is not causing the jumping. I've also tried adding blocking to the input to only accept updates when the cursor is redrawn in an attempt to keep it more consistent but see no significant change. 

    #ifdef MODE1
    if(configMode&&(MwRcData[YAWSTICK]>MAXSTICK)) // MOVE RIGHT
    #else
    if(configMode&&(MwRcData[ROLLSTICK]>MAXSTICK)) // MOVE RIGHT
    #endif
    {
    if(hasDraw){
    waitStick = 1;
    COL++;
    if (COL>3) COL=3;
    hasDraw = 0;
    }    
        
    }

jflyper, 2016-10-27:
Are you experiencing this with F4 targets?

HaseebNain, 2016-10-27:
Yes it is an F4, that's interesting. 

I'll look into it, it definitely seems to relate to the speed of the UART off of the F4 and the communication between the ATMEGA and F4. 

Thanks for linking the betaflight thread, can't believe I missed that.

ShikOfTheRa, 2016-10-28:
Not experienced myself, but I don't have an F4. Nor an F3 anymore :( ...
Maybe F4 serial much faster?

There is a menudir/oldmenudir which was added at request of people wanting to adjust PIDS, settings etc. rapidly. it works by increasing the step range by 1 until hit constraint of 5. So in going for 0-say 1000 it does it in 1/5th of the time.

If its related - could try all instances of constrain(menudir,-5,5) to constrain(menudir,-1,1) to test

HaseebNain, 2016-11-02:
I did try removing the ability to increase the step and saw the cursor still jump randomly.

I have tried the different constraints and see no significant impact on the cursor behavior. 

I am thinking about hooking an arduino up to the F4 to see how the data sent to the Atmel looks, maybe there is some issue with how the OSD firmware reads the F4 information?

jflyper, 2016-11-02:
> I am thinking about hooking an arduino up to the F4 to see how the data sent to the Atmel looks, maybe there is some issue with how the OSD firmware reads the F4 information?

I don't think so.

What FC and MWOSD is doing in terms of stick movement is that the MWOSD is periodically sending a request for RC channel data and FC responding with the RC channel data. If something is wrong with the MWOSD's processing, it would happen anytime as you move the stick around.

As you mentioned that the FC freeze when disconnected from the configurator, I would suspect something wrong with MSP processing on FC in regards to connect/disconnect status of the configurator.

One that bit me was this one:
Look Ma! My VCP port is a zombie! (betaflight/betaflight#1432, and [comment in betaflight/betaflight#1422](https://github.com/betaflight/betaflight/pull/1422#issuecomment-257166195)
There might be others.

HaseebNain, 2016-11-02:
Ok that's good to know! 

I wonder why the stick movement seems so sporadic then. I have tried putting blocks into the firmware after an input is sampled and noticed that it does not jump as often but still has inconsistent jumps.

I have also placed "if" statements around the cursor updating to only take information when the cursor has been updated and noticed an increase in consistency.

for example in "serial.c"

    #ifdef MODE1
      if(configMode&&(MwRcData[YAWSTICK]>MAXSTICK)) // MOVE RIGHT
    #else
      if(configMode&&(MwRcData[ROLLSTICK]>MAXSTICK)) // MOVE RIGHT
    #endif
      {
        if(hasDraw){
        waitStick = 1;
        COL++;
   
       if(COL>3) COL=3;
        hasDraw = 0;
       }    
     }

hasDraw is set equal to "1" in "Screen.ino" at the end of the displayCursor() function.

HaseebNain, 2016-11-02:
In the case of Betaflight I don't see any issue with the serialization of RC data

    case MSP_RC:
        headSerialReply(2 * rxRuntimeConfig.channelCount);
        for (i = 0; i < rxRuntimeConfig.channelCount; i++)
            serialize16(rcData[i]);
        break;

ShikOfTheRa, 2016-11-03:
Are you using PPM? I have seen a number of badly set up systems as exceed PPM frame.

Or perhaps maybe there is some sort of timing issue... e.g rcData[i] changing due to interrupt mid serialization. as its worse with F4 it seems that is most likely to be related to BF side.

Actually, I have just added debug data display screen which includes the RC values. Might be worth running that for a while to see if fluctuations from rcData.

If I were starting from scratch with MWOSD I probably would do some atomic blocking for some of the 32 bit manipulation as I have seen the effects..., but as OSD is non critical to actual flight and I haven't seem lockups I'm not going to invest the lot of time and analysis needed to fix an issue that no-one seems aware of!

HaseebNain, 2016-11-03:
I am using SERIAL_RX for radio communication.

I also think it is an issue with rcData[] and how it gets the information. I've looked into betaflight and did not see anything to strange. I do not think adding blocking calls into betaflight will be too good of an idea, don't want to lock up radio communication and cause a crash haha.

what do you mean by atomic blocking? The way the data is currently accessed seems to be working fine in the OSD firmware. 

I think if we start cutting the rcData[] into peices we could miss important stick information and lead to more instability.

ShikOfTheRa, 2016-11-04:
I'd be inclined to put debug screen on and see what data shows up on rcdata. I have many times seen ppm issue - that for sure is real and common.

Not to confuse the atomic block with this issue - and just as an FYI.. Below is a simple descriptions/demonstrations of atomic issues on 32 bit processes... I have no idea if its a real issue on MWOSD. I saw some possible evidence last year which only showed up in fast simulation... not real normal operation. Something to be aware of..

https://forum.arduino.cc/index.php?topic=73838.0

HaseebNain, 2016-11-10:
It seems that the information is not the issue.

I have attempted different speeds, assuming that the F4 is simply too fast for the ATMEGA but saw no improvement over the different preset MSP speeds in config.h

I am wondering if it is worth reworking the MSP in the OSD firmware, perhaps constantly taking in MSP_RC information as it does now, but once the config menu is enabled, only taking MSP_RC data after each action (ie. cursor drawing)

Hazakins, 2016-11-10:
over on the betaflight git they recommended 
"Have you tried changing the serial task priority and period?

Try in fc_tasks.c try setting staticPriority for TASK_SERIAL to TASK_PRIORITY_MEDIUM and desiredPeriod = 1000000 / 200. I'm not proposing that as the final solution, but changing it will show if it is a priority issue."

Hazakins, 2016-11-10:
Hope you guys figure this out. Got a Revo running scarab-osd and its pretty wonky

jflyper, 2016-11-11:
Rcdata from fc is not just flowing. They are passed to osd as they are requested by the osd. Request-response based protocol running there.

So, the choppiness has nothing to do with processor's speed on both sides.

jflyper, 2016-11-11:
I will start looking into this problem after Sunday evening (as posted in the BF thread).

jflyper, 2016-11-14:
It turned out that F4 UART is broken in the way that it intermittently transmit batch of garbage on to the MSP response stream. See this [comment](https://github.com/betaflight/betaflight/issues/1386#issuecomment-260205776) in betaflight/betaflight#1386 for details.

I think we can close this one on this side.

HaseebNain, 2016-11-14:
Yea I saw that, it seems everything on the OSD side is working correctly and this stems to be an issue with the F4 itself.

Will close this for now.

ShikOfTheRa, 2016-11-17:
Good troubleshooting....

HaseebNain, 2016-12-02:
I'm reopening the issue as I think that MSP needs to be looked at on this side as well. 

I have the OSD stop responding all together, stating "NO DATA" when in the menu. It was solved through a complete reflash and has not had an issue since then 2:34pm MST (12/1/2016). 

I am going to look into the MSP handling, perhaps some of the corrupt information from the F4 is causing a system failure?

Or some memory leak, which could explain why a reflash of the firmware causes it to work for the time being?

ShikOfTheRa, 2017-01-10:
Still think there is an issuue? I can't see / replicate anything

HaseebNain, 2017-01-10:
No, I think the issue has been cleared. 

I still get the occasional "NO DATA" but do not think it is related to MWOSD rather information may be corrupted when sent from the F4, increasing the timeout from 3 seconds to 5 or 6 might help but I'm not so sure.

closed 2017-01-10 by ShikOfTheRa