ShikOfTheRa/scarab-osd #327: FuriousFPV Innova OSD-VTX support

jflyper, 2016-10-27:
For discussion and info exchange on support for FuriousFPV's Innova VTX support.

The Innova basically seems like M328P driving MAX7456 and FX758-2. The vendor modified (and not yet disclosed) version of MWOSD (of unknown release) is used.

Estimated changes are as follow.
1. Bit-banging SPI-ish controlling FX758-2 through 3 (or 2) digital output pins.
2. Config menu extension which thoughtlessly squeezed 3 more rows for vtx parameters into PID page (and made the product incompatible with NTSC).

Product page:
[INNOVA OSD-VTX board 20x20mm](http://furiousfpv.com/product_info.php?products_id=76)

RCG thread (PIKO BLX):
[PIKO BLX Micro Flight Controller - Everything Changes](https://www.rcgroups.com/forums/showthread.php?t=2664875&page=75#post36053035)

FX758-2:
[https://szfxt.en.alibaba.com/product/60303419485-214503132/FXT_Lightweight_5_8GHz_200mW_VTX_FX758_Support_SPI_AV_Transmitter_Module_for_Raceband_VTX.html](https://szfxt.en.alibaba.com/product/60303419485-214503132/FXT_Lightweight_5_8GHz_200mW_VTX_FX758_Support_SPI_AV_Transmitter_Module_for_Raceband_VTX.html)
[http://www.banggood.com/FPV-5_8G-200mW-32CH-Wireless-Audio-Video-AV-Transmitter-Module-FX758-2-p-980317.html](http://www.banggood.com/FPV-5_8G-200mW-32CH-Wireless-Audio-Video-AV-Transmitter-Module-FX758-2-p-980317.html)

jflyper, 2016-10-27:
Related
#315 Smart audio support
#323 Where is the code for controlling a 6705 5.8ghz video transmitter?

#273 add support for BetaFlight custom MSP commands
#325 Canvas mode support for BF-CMS and inverted char support
betaflight/#1352 Look Ma! BF config menu runs on my Minim!
betaflight/#1283 Add TBS SmartAudio support

#321 Support for Furious FPV Piggy OSD (Not technically related)

jflyper, 2016-10-27:
Since the bit-banged SPI for MAX7456 is order of several magnitudes easier than non-blocking software serial, I would say it's a weekend job to add a RTC6705 driver, and another to add a menu page, once started. (Adding a menu page would take more time... sigh.)

sonicxx, 2016-10-27:
There's also this:

https://github.com/sblakemore/cleanflight/commit/2d69dfd79df4f4fc4aa0aa9604e728f96c0231d2 Sorry if it's not reference properly, I'm a neophyte.

jflyper, 2016-10-27:
@sonicxx Yeap. I've worked with him for a few weeks to integrate my bit-banged version with his stick control: cleanflight#1948

jflyper, 2016-10-27:
So, we should at least find out which FX758-2 pin is connected to which M328P pins.
[image]

ShikOfTheRa, 2016-10-28:
The impulse source is here...

https://github.com/PickwickGroup/scarab-osd/tree/impulserc_vtx

jflyper, 2016-10-28:
Ah, then it's MCU-less version of the SM21xT family.

ShikOfTheRa, 2016-10-28:
Interested in adding - or shall I see if the original dev can add in support?

jflyper, 2016-10-28:
Furious is said to upload and disclose their source for innova, which I believe (at least) functionally equivalent. Let's wait.

ShikOfTheRa, 2016-10-28:
Actually I thought they were the same. My mistake. 
Innova != Impulserc Helix

But they might be functionally the same. 
Would be a nice addition.

sniperxnl, 2016-10-30:
Can i do a firmware update on the novva form Furious with passtrough from fc?

jflyper, 2016-10-31:
I don't have Innova, but in theory, yes.

However, updating Innova with current MWOSD will loose the VTX control capability.
This issue is for discussion toward the support of this capability.

Rclf, 2016-12-02:
I was configuring the innova in mwosd and I accidentally pressed default . I can still see the osd in mwosd gui but the video is gone, it disabled the vtx part somehow

jflyper, 2016-12-12:
@Rclf Did you try specifying NTSC/PAL in GUI?

jflyper, 2016-12-12:
So, FuriousFPV is coming out with V2 of Innova. Any news on source code disclosure?

Rclf, 2016-12-12:
Yes of course, this is not it.  My guess is that somehow it messed the firmware.

jflyper, 2016-12-12:
If you've decided to stash it, can you do some investigation mentioned [above](https://github.com/ShikOfTheRa/scarab-osd/issues/327#issuecomment-256545317) ?

ShikOfTheRa, 2016-12-12:
@rclf - suggest to ask your supplier to provide the source code - we can help you out then.

proggod, 2016-12-12:
Has anyone attempted to contact them and request it?

ShikOfTheRa, 2016-12-12:
Yes - I have and they said they would. I haven't received it yet though

proggod, 2016-12-12:
How long has it been?

ShikOfTheRa, 2016-12-12:
@proggod 2 months maybe. I have just emailed again.

Rclf, 2016-12-12:
> I'll talk to my contact for you

I am waiting for the firmware as well.  Should have it in a couple of days.

ShikOfTheRa, 2016-12-13:
Had a mail from furious this morning - he is chasing the dev

proggod, 2016-12-13:
Great, I asked someone to light a fire under them to get it to you :)

TDIPWR, 2016-12-22:
Any word on the firmware?

Rclf, 2016-12-22:
No not yet,  I emailed again yesterday  still no answer.

DaveLister007, 2017-01-11:
I'm gonna push this little further - any news on source code from vendor?

If not, how different do you reckon this vtx is from this one??
https://github.com/PickwickGroup/scarab-osd/commit/9cb6e99da3446d763c82024485a4e6e6e0e7bd5e

Quick beeping with my multimeter revealed that CH1-CH3 pins of FX758 are connected via 10K resistors to MCU pins, so I'm guessing SPI. There are probably no dedicated pins for power settings, so it's probably set via SPI.

jflyper, 2017-01-14:
@DaveLister007 
> If not, how different do you reckon this vtx is from this one??

Not much. You can probably come up with a working code only by substituting pins for spi to match that of innova.

DaveLister007, 2017-01-19:
I messaged FuriousFPV and got the source code into email. There is nothing like sending something via serial port/SPI added, just raped PID menu for setting channel, and storing/loading it from EEPROM. Either im blind/dumb, setting channel/power is done via some other magic outsice MCU, or they deleted that piece of code before sending it to me. If anyone wants to have a look, let me know, i'll send it to you.

ShikOfTheRa, 2017-01-19:
yeah. like to take a look. I'm hoping to add vtx support into mainstream code. If don't hear back from them its probably not going to be compatible.

DaveLister007, 2017-01-19:
This is what I received. Seems to be based of 1.6 version. Rename it to .rar (githib doesn't allow that extension).
[FIRMWAVE.txt](https://github.com/ShikOfTheRa/scarab-osd/files/718014/FIRMWAVE.txt)

u8mydog, 2017-01-19:
Try diffing the two OSD dirs. For example, globalVariables.h has EEPROM_ADR_BAND, EEPROM_ADR_CHANNEL, EEPROM_ADR_FREQ and EEPROM_ADR_POWER. Looks like serial.ino uses the eeprom values via PID settings.

ShikOfTheRa, 2017-01-20:
Does appear that this does not include all the code..

Good to see the Piggy is stock to confirm what thought

jflyper, 2017-01-20:
+1 This is a deteriorated code.

DaveLister007, 2017-01-23:
So I messaged them again, and received some this. This one looks much better.
[MW_OSD_VTX.txt](https://github.com/ShikOfTheRa/scarab-osd/files/724622/MW_OSD_VTX.txt)

ShikOfTheRa, 2017-01-23:
Well that took some time. It looks about right.

Can now consider which to implement. If any!

Not going to add anything new as closed doors for anything new for 1.7 . So maybe this is for 1.7.1

Thanks for the files. Hugely appreciated

jflyper, 2017-01-23:
```
#define SPI_DATA_PIN    10
#define SPI_SELECT_PIN    9
#define SPI_CLOCK_PIN   8
```
This is what I wanted to know. There's nothing else special (except for hard coded EEPROM addresses).

And I've noticed that tx5823-pro code is used also.
[tx5823-pro](https://github.com/sheaivey/tx5823-pro/blob/master/src/tx5823-pro/tx5823-pro.ino)

jflyper, 2017-01-23:
I'm now interested in how to control peripheral devices on OSD from FC...

Extend MSP_OSD message?

DaveLister007, 2017-01-23:
@jflyper If the pins were only thing you were interested in, I could have told you month ago by reverse engineering it :)

Anyway, canvas mode and commands to set channels/power via FC is way to go IMHO.

jflyper, 2017-01-24:
@DaveLister007 
It has been requested in the [6th comment](https://github.com/ShikOfTheRa/scarab-osd/issues/327#issuecomment-256545317)  above ;)

HenryIvarsson, 2017-02-10:
Thank you guys for all the hard work! I flashed my BG Innova clone to the latest mwosd beta without thinking, should I try flashing back to the version posted above or wait for MWOSD support? When do you think there can be a version to test?

jflyper, 2017-02-10:
@HenryIvarsson The current MWOSD doesn't support Innova's VTX capability. You should flash it back. The support schedule is not available yet (Probably after March 1st, when Betaflight 3.1.6 patch is released.)

jflyper, 2017-02-12:
Okay, I can probably spend next couple of days for the initial cut of the Innova support.

HenryIvarsson, 2017-02-12:
That sounds awesome, thank you!

ShikOfTheRa, 2017-02-12:
@jflyPer - I have Vtx branch ready to merge for another  one. Can you wait until I merge? Might be a lot of common code can reuse. Need to review on my pic should be tomorrow or day after at latest

jflyper, 2017-02-12:
Dah! I already did most of the code. Well, I'll pause as it is right now.

HenryIvarsson, 2017-02-13:
Nice!! Is there any way to download a compiled version to test?

HenryIvarsson, 2017-02-13:
Ah, sorry, missed the ShikOfTheRa's comment above, no stress guys! You guys rock!

ShikOfTheRa, 2017-02-13:
@jflyper Dah! 2
Just synced a branch. Does this work for you - or possible to include its elements into your stuff. 

Used ImpulseRC code as base.

Menu separated from hardware type – to allow multiple hardware implementations

VTXHARDWARETYPE.ino will need: following and any associated code
  vtx_init() // set up hardware
  vtx_save()  // save vtx settings
  vtx_read() // load vtx settings from eeprom
  vtx_set_frequency(vtxBand, vtxChannel)  to change 

def.h
#ifdef VTX_RTC6705  | (add HW here) for some other hardware to enable menu

following settings are used:
  Settings[S_VTX_POWER]   = vtxPower;
  Settings[S_VTX_BAND]    = vtxBand;
  Settings[S_VTX_CHANNEL] = vtxChannel;

Its untested on hardware, so prob need a little debugging…

macfly1202, 2017-02-16:
HI, I've got a genuine Innova vtx V1. It's perfectly working. But Essential issue I got is my cam is NTSC and innova - actual code - only support PAL so I can't see last line > impossible to navigate between menu in this case...I'm not alone in this case according to RCGroup thread. I'm sure there is solution on this problem like enumerate new page OR changing font in worst case (wendy3x5.h could do the trick, even if partial for some letters)  
So if needed 'tester' of alpha, beta version for Innova, I could made a try. I see the new VTX branch but doesn't know if could be test right now.

I've got a question : Does somebody have already try to "plug externally" micro or minimOSD via SPI wires like we do on this project https://github.com/markohoepken/rx5808_pro_osd/wiki/wireing-diagram. Pin are quiet easily accessible with soldering and don't understand why not using it instead of MSP ?

jflyper, 2017-02-16:
@macfly1202 
The VTX branch is created without an actual hardware, so need some testing/debugging. And it's for ImpulseRC's Helix OSD, so must still be modified to tailer for Innova.

> I've got a question : Does somebody have already try to "plug externally" micro or minimOSD via SPI wires like we do on this project https://github.com/markohoepken/rx5808_pro_osd/wiki/wireing-diagram. (it's perfectly working on my HeadplayHD). Pin are quiet easily accessible with soldering and don't understand why not using it instead of MSP ?

I'm confused by this question.

- If I understand the link you've provided correctly, it's for receiver side configuration.
- In the wiring, SPI is controlling MAX7456 and RX5808, controlling devices.
- MSP (and other serial protocols) are used to communicate with data source (e.g. FC), to receive what to display as overlay.

And what do you mean by "Using SPI instead of MSP"?

macfly1202, 2017-02-16:
@jflyper  Thanks for clarification about VTX branch.
About 
I'm sorry If not very clear. I'm not developper just DIYer. If I well understand, Omnibus F4 pro or betaflight F3 board display infos receive (and transmit setup) with SPI connection instead of traditionnal "serial" uart connection. In my example link, SPI from 328p is use to display and change channel of rx5808. But I was simply asking instead of wiring micro/minimOSD with uart connection why not replacing UART with  SPI wiring from FC to 7456 with multiples advantages (freing 1 uart & DMA on >=F3 & cpu).
Perhaps it's a nonsense for you but simply asking. Thank you very much for all the great work you done.

jflyper, 2017-02-16:
FC-integrated MAX7456 boards use SPI to control the MAX7456.

FC-integrated MimimOSD-variant use UART to communicate with M328P, M328P use SPI to control the MAX7456.

So, what are you proposing?

FC-integrated MinimOSD-variant to use SPI to communicate with M328P?

macfly1202, 2017-02-16:
NOT integrated FC use SPI to control the MAX7456 in 1st time. Why must it be integrated ? 3-4 wires to solder between FC F PIN and MAX PIN.   
In 2nd time, FC use SPI to control RX.

u8mydog, 2017-02-16:
I have wondered the same thing if I understand the original question.
Give minimOSD the option to enable FC integration by using SPI communication from FC -> M328P -> MAX7456. 

Basically, the M328P acts as a simple bridge to the MAX7456 moving all OSD functionality to the FC instead of the M328P.

HenryIvarsson, 2017-02-17:
Hey guys, any code ready that I should test on my Innova yet? If so I might need som help compiling if it is not posted somewhere already.

jflyper, 2017-02-17:
@HenryIvarsson [Read this](https://github.com/ShikOfTheRa/scarab-osd/issues/327#issuecomment-280391069)

jflyper, 2017-02-17:
@u8mydog 
> Basically, the M328P acts as a simple bridge to the MAX7456 moving all OSD functionality to the FC instead of the M328P.

If we do this via SPI, we don't need M328 at all.
If we do this via UART, it is too slow, at least atm. We have tested this option during the development of the CMS over MSP_DISPLAYPORT; too much overhead sending characters instead of sending binary data.

HenryIvarsson, 2017-02-17:
@jflyper Thank you, just so I understand correctly, I assume I cannot test Innova on the Impulse settings on the vtx branch as the hardware is to far apart?

[100 more replies]

closed 2018-01-10 by ShikOfTheRa