ShikOfTheRa/scarab-osd #328: Enabling AIRBOTMICRO23 produces blank screen

jflyper, 2016-10-30:
The issue was brought up in the fpvlab
[http://fpvlab.com/forums/showthread.php?34250-MWOSD-for-MULTIWII-NAZE32-BASEFLIGHT-HARIKIRI/page317](http://fpvlab.com/forums/showthread.php?34250-MWOSD-for-MULTIWII-NAZE32-BASEFLIGHT-HARIKIRI/page317)

> getting this new microosd from myairbot http://shop.myairbot.com/index.php/microosdv23.html got me back on MWOSD, you guys do an amazing job there!!!

in 1.6.4 you have introduced a respective hardware-definition in config.h already. it just don't seem to give any video output (no overlay, no cam signal) if i define "airbot microosd 2.3", it does work like a charm with standard "minimosd" defined though. i wonder if i'm missing some other necessary settings here? 

on a sidenote, this hardware seems to exclusively work with your firmware, couldn't get it to output any video signal on minimosd standard, minimosd-extra or anything else but MWOSD.

ShikOfTheRa, 2016-10-31:
I have a board that looks real similar. Labelled microsod v2. This runs a max chip...
Anyway, tested on my new ab7456 board and I recommend leaving MAX7456HWRESET enabled when MAX_SOFTRESET enabled.

Reason: If reset isn't connected, it makes no difference.. If it is connected then we get hardware reset and correct initialisation of reset line too.

I can confirm that it works on my board with softreset enabled and hardware enable at same time. This has the reset line connected. If hardreset not enabled, the line is left low by power on defaults.

I suspect the board he has isn't same as yours... might be worth checking the chip.

ShikOfTheRa, 2016-11-25:
Think I have the answer now...

Any boards prior to 2.35 uses a MAX7456 chip.
If define AIRBOTMICRO23 and board is less than 2.35 you will get a blank screen as HW reset line isn't initialized. As above recommend to leave HW reset initialization in place as addition to soft reset. Future boards likely to have HW reset line in place too.

closed 2016-11-25 by ShikOfTheRa