ShikOfTheRa/scarab-osd #329: Innova OSD GPL Violation

btreecat, 2016-11-08:
It seems that with Furious FPV using some sort of derivative of MWOSD. They do not however provide their changes as required by the GPL, which mwosd/scarab-osd is licensed under. 

I think we should start contacting them regarding the release of their modified source. This could make integrating vtx changing easier for other hardware as well.

ShikOfTheRa, 2016-11-09:
They are  preparing code for release.

Note broadly equivalent source is available here:
https://github.com/PickwickGroup/scarab-osd/tree/impulserc_vtx

btreecat, 2016-11-09:
Yes they emailed me back finally letting me know they are interested in
complying with the licens

ShikOfTheRa, 2016-11-25:
Closing this and updates will be in #327

closed 2016-11-25 by ShikOfTheRa