ShikOfTheRa/scarab-osd #330: KISS compile does not update mAh

RBeaubien, 2016-11-08:
When using current sensor connected to OSD, mAh used does not update when compiled for KISS FC.  This is in R1.6.2.  The current sensor is correctly reporting.

ShikOfTheRa, 2016-11-09:
For KISS:
If “Use FC” is enabled, it expects to get Amperage from the KISS ESC

I just tested on 1.6.5 with Display amps enabled, Display mah enabled, use virtual disabled, use FC disabled and it works Ok for me. Current sensor connected direct to OSD. The code looks to have not changed between those versions either.

RBeaubien, 2016-11-09:
"Use FC" is NOT enabled.  I want the current sensor connected to the OSD to be used.

RBeaubien, 2016-11-14:
I think I found the issue.  I did not have the 5v Reference turned on.

ShikOfTheRa, 2016-11-17:
OK great

closed 2016-11-17 by ShikOfTheRa