ShikOfTheRa/scarab-osd #331: Use FC for Voltage and Current readings, default not working?

HaseebNain, 2016-11-09:
I am trying to understand the communication between the FC and the OSD. 

In the config.h file I defined "USE_FC_VOLTS_CONFIG" however the OSD does not report the value of voltage and current. 

Voltage only seems to be enabled through the OSD itself, when it is turned on using the stick commands.

Both work fine when enabled through the OSD_GUI.

ShikOfTheRa, 2016-11-10:
Not used myself, but I think AUTOCELL needs to be enabled. Also Use FC voltage...
Perhaps documentation / implementation needs a little tweak!

These two with "Use FC voltage enabled seem to be whats required 
#define AUTOCELL                  // Uncomment this to use varying cell count batteries. Overrides GUI/OSD voltage warning settings. Uses CELL_VOLTS_* below unless AUTOCELL_ALARM defined
#define FC_VOLTAGE_CONFIG         // Additionally uncomment this if you want to use the vbat voltage config with BASEFLIGHT and CLEANFLIGHT on the flight controller (include: min cell voltage, max cell voltage and warning cell voltage)

HaseebNain, 2016-11-10:
It doesn't seem to be enabling the FC by default. 

I enable it through the OSD and then when it works, wipe the EEPROM and reflash it with those configs enabled.

HaseebNain, 2016-11-10:
@ShikOfTheRa 

So I went in and used the code for the GUI and was able to enable these as defaults by changing values in the default array (EEPROM_DEFAULT) in GlobalVariables.h

I can look into creating a quick function and add some more parameters to config.h so that these values can be set manually. 

Just adding a Settings[S_MWAMPERAGE] = 1 and a Settings[S_MAINVOLTAGE_VBAT] = 1,
into MW_OSD.ino should be enough.

ShikOfTheRa, 2016-11-17:
yes could force Settings[S_MAINVOLTAGE_VBAT] = 1 if selecting FC_VOLTAGE_CONFIG 

AMPERAGE - no

ShikOfTheRa, 2016-12-21:
OK - am told of scenario where can't implement - they want to use config from FC, but use the analog.

It means this will have to be manually set. can't implement further.

closed 2016-12-21 by ShikOfTheRa