ShikOfTheRa/scarab-osd #335: Incompatibility with Arduino 1.6.10 and later

jflyper, 2016-11-16:
Arduino 1.6.10 and later decided to go strict on forward declarations, or effectively doing the same thing.

Related: #334 #332

ShikOfTheRa, 2016-11-17:
great. I see in reference it compiles OK for you n mac though? I am travelling so not enough broadband allowance to download and try myself. Will be a couple of weeks before can look at

jflyper, 2016-11-17:
@ShikOfTheRa 
I think I was fooled by multiple arduino instances when I did that testing for #334.

The issue is present and consistent among 1.6.10 and later in my environment.

jflyper, 2016-11-18:
So, does anybody know what has happened in 1.6.10?

Looks like Arduino preprocessing (merge all ino, insert some includes and collect declarations) is corrupt or has changed in some way.

ShikOfTheRa, 2016-11-18:
Weird. 

I installed 1.6.12 and clicked the remove old version .option Rebooted and can compile OK with no errors. Just also tried a manual delete and reinstall no issues.

Cant find any other versions of arduino on my pc and only 1 version of avrgcc which is  4.9.2

ShikOfTheRa, 2016-11-21:
Can you confirm if the latest compiles?

jflyper, 2016-11-23:
It compiles! What did you do?

sppnk, 2016-11-24:
May be it has something to do with [this weird arduino ide](https://github.com/arduino/Arduino/issues/2989) issue?

ShikOfTheRa, 2016-11-24:
I think there must have been some IDE changes with max characters on a single line so I simply split the fontdata line into 2.

It took some finding......

jflyper, 2016-11-24:
@ShikOfTheRa Thanks for your findings! Closing.

closed 2016-11-24 by jflyper