ShikOfTheRa/scarab-osd #337: Any plans to support MAVLink version 3?

anton-matosov, 2016-11-25:
Pixhawk and APM are running MAVLink v3 now, but MWOSD supports only legazy v1 (at least as per config.h).
Currently when I connect Pixracer with px4 stack on it OSD breaks - it shows all the font characters instead of the data.

ShikOfTheRa, 2016-11-27:
Is there a v3 ?? I just looked at mavlink on github and I can only see reference to 1 and 2 libraries. But I do not know apm/pixhak stuff very well
I guess will have to but I'll have to learn v2 first.
Due to memory, MWOSD uses its own mavlink protocol parser. I'll have to create one to suit this. Unfortunately not going to be quick for me to do this

anton-matosov, 2016-11-28:
http://mavlink.org/messages/common
> ### MAVLink Protocol Version
> This file has protocol version: 3. The version numbers range from 1-255.
So MAVLink protocol already has version 3, but I have checked current master of px4 and they still use v2.0

ShikOfTheRa, 2016-11-30:
yeah I saw that before, but I still can't find a v3... I don't think it exists.
In libraries I can only see a v2 and that looks pretty recent. 

In docs its says you should be able to send at v1 for compatibility

2black0, 2016-12-03:
have same problem, while osd connected with apm (tx rx) it show all characters and random reboot, after unplug tx rx, it show but no data (definitely yes cause no data input for OSD)

hope this can fix, :)

2black0, 2016-12-14:
I check on my apm 2.6 (with firmware arduplane v3.4.0) it use mavlink v3

ShikOfTheRa, 2016-12-21:
Hi - I have found a bug in the mavlink code - this would cause screen full of garbage in some mode scenarios.

Welcome to try the latest and feedback...! 

https://github.com/ShikOfTheRa/scarab-osd/archive/master.zip

ShikOfTheRa, 2017-01-05:
@anton-matosov 
@2black0 
Can someone try this. Today I have spent long time understanding mavlink protocol. Made a number of changes which I think may fix this. I have sent hundreds of thousands of mavlink packets form various sources and it seems stable and working now.

The only thing I am unsure of is armed status. The test data I have seems inconsistent with the documentation.

anton-matosov, 2017-01-05:
I can try this out over the weekend

2black0, 2017-01-07:
Will try it maybe in next week, cause I need to build apm system again after the old one was sold  :D

ShikOfTheRa, 2017-01-07:
Thanks both.

ShikOfTheRa, 2017-01-22:
Feedback is now 100% so closing. Thanks. There was an issue and it appears resolved

closed 2017-01-22 by ShikOfTheRa

ShikOfTheRa, 2017-08-04:
For PX4 - there is also now support for this