ShikOfTheRa/scarab-osd #339: Add settings for AHI sensitivity

nhadrian, 2016-12-09:
Hi,

I'd like to ask setting for roll and pitch AHI sensitivity in config.h since different cameras and optics gives different POV. 

Thanks in advance.

Regards Adrian

ShikOfTheRa, 2016-12-09:
OK - added this in config.h.

I gave it a quick test and it looked OK on the bench.
These are now in config.h:

100=normal so consider 100%
75 would reduce pirch/roll by three quarters 

//#define AHIPITCHSCALE 100          // Specify scaling sensitvity for Pitch. Higher number = pitches more on OSD         
//#define AHIROLLSCALE 100           // Specify scaling sensitvity for Roll. Higher number = rolls more on OSD

ShikOfTheRa, 2016-12-12:
@nhadrian lt me know how testing goes. Cheers

nhadrian, 2016-12-18:
Hi, itt looks like scale is working.
But I found another issue, after raching an angle of around 40...45 deg, AH flips to the opposite direction (like copter is up-side down) even if AH limits are set to 600-700 range.

ShikOfTheRa, 2016-12-21:
OK - will have a look.

The scaling is purely for scaling on the display. Real values should not have changed.

jflyper, 2017-01-02:
I think the AHI flipping is the normal behavior.

Ref: AH behavior on extreme pitch #291

When pitch increases, roll angle will gradually loose it's significance. Think of the extreme case when a vehicle is up right; there's no left or right. And this happens to attitude values sent by FC; there is no way the OSD code can compensate the effect (even by scaling).

You can reproduce the behavior with configurator's attitude indicator graphics, too.

ShikOfTheRa, 2017-01-10:
@nhadrian I don't think anything more can do here. In some ways, it looks like the FC is sending incorrect pitch/roll info. I would always expect the OSD AHI to broadly represent the horizon.

Will leave open for a week or two for any further comment / feedback

closed 2017-01-22 by ShikOfTheRa