ShikOfTheRa/scarab-osd #340: A7105 chip RSSI Support

2black0, 2016-12-14:
we know, there are many receiver based on A7105 Chip like Flysky FS-i6 / FS-i6B / or the older like TGY 9X, the chip have RSSI pin out (based on datasheet) but unfortunatelly the RX dont have feature for RSSI Pin out,

some people do hack directly get data RSSI from Pin 1 of A7105 and read it use uC (atmel / arduino) and generate RSSI pwm / analog data

sorry to request this, can MWOSD supported to directly get data from RSSI pin1 A7105
I attach some document for reference

A7105 supports 8-bits digital RSSI to detect RF signal strength. RSSI value is stored in ADC [7:0] (1Dh). Fig 17.1 shows a typical plot of RSSI reading as a function of input power. This curve is base on the current gain setting of A7105 reference code. A7105 automatically averages 8-times ADC conversion a RSSI measurement until A7105 exits RX mode. Therefore, each RSSI measuring time is ( 8 x 20 x FADC). For quick RSSI measurement, recommend to set FSARS = 1 (FADC =8MHz, 20 us measuring time). For power saving, recommend to set FSARS = 0 (FADC =4MHz, 40 us measuring time). Be aware RSSI accuracy is about ± 6dBm. 

[A7105 Datasheet v1.4.pdf](https://github.com/ShikOfTheRa/scarab-osd/files/653298/A7105.Datasheet.v1.4.pdf)
[image]

ShikOfTheRa, 2016-12-21:
Had a quick look at this. A lot of work and quite a hardware hack / very fine soldering from looks of it.
Sorry but can't see this being supported.

jflyper, 2016-12-25:
I think the pin #1 RSSI has nothing to do with the above mentioned "17.1 RSSI Measurement" things.

Isn't it just an analog voltage? If so, it can be connected to RSSI input of MinimOSD. Probably need a small capacitor to smooth out the small changes.

ShikOfTheRa, 2016-12-26:
ah yeah - looked again. Think your right. 
17.1 is more for the microp operating the ic.

ShikOfTheRa, 2017-01-10:
I think answer is above.

Nothing more can do here

closed 2017-01-10 by ShikOfTheRa