ShikOfTheRa/scarab-osd #345: Horizon Bar is showing  LON LON LON &~ LON LON LON

patrickpoirier51, 2016-12-29:
Running CleanFlight latest release on a F3 Based FC.
Just uploaded version MWOSD R1.6 with Standard Config.h on a Mini OSD.

Problem:
Looking and the Horizon , it is not showing bars and dots but:
LON LON LON &~ LON LON LON
And this is sequentially displaying (depending on the angle):
LON - LAT - MAH -KMH - KNTH -MS -THR -VEL -ALT

Never used this software before, might have missed something ? Please advise

advancetom, 2016-12-29:
U must use a 5Volt Programmer!  got the same Problem

patrickpoirier51, 2016-12-29:
Thanks for the tip , I tried another Programmer and its OK now :-)

patrickpoirier51, 2016-12-29:
Might be interesting to add this on the install guide ?
Have an FTDI breakout board (which usually connects to a USB port)  with **sufficient power (5 Volts) and** the proper headers and to connect to your OSD hardware.

jflyper, 2016-12-30:
The requirement for the good 5V source when flashing (and probably writing EEPROM) is known for a long time, but yes, it's not mentioned anywhere in the DOCUMENTATION.

closed 2017-01-10 by ShikOfTheRa