ShikOfTheRa/scarab-osd #346: MWOSD GUI Mac OS X 10.11

dan13373, 2016-12-30:
Hi Shik,

I have tested the micro minimOSD with an FTDI to USB converter on my Mac with the correct drivers (dev/usb.serial is the port). The Arduino loads the MWOSD.ino firmware on the minimOSD successful without an error. 

I then restarted my Mac and then started the MWOSD GUI by disabling the Gatekeeper in order to make it start. I set the BAUD Rate to 57600 (as the Arduino IDE was using) and tried to connect to the usb.erial (stated in the GUI like that)...but it keeps saying "Waiting for OSD". I checked the wiring - but I guess this must be correct since the upload of the firmware was successful.

Is there a chance to check/set the Port Setting like in Arduino IDE?

Best,
Dan

jflyper, 2016-12-30:
Default speed is 115200. Try it.

Arduino IDE 's speed is for boot loader. MWOSD as a standalone software has it's own speed, which defaults to 115200 unless you tweak it in Config.h.

dan13373, 2016-12-30:
Not working :( but thx

ShikOfTheRa, 2016-12-30:
@dan13373 I'm not clear from comments- did you change GUI baud to match the OSD? Its not obvious - have to click on com settings button to show baud rate option.

dan13373, 2016-12-30:
Yes I have tried that.
How long to wait until the GUI should respond?

dan13373, 2016-12-30:
Edit:
I changed to match OSD and edited the config

ShikOfTheRa, 2016-12-30:
Less than 10 secs. enough time for any reboot and startup delays in the code..

just a thought I get this sometimes with OSDDOGE / BROSD style boards when teh FC is connected and I forget to disconnect. I done it today... several times!

dan13373, 2016-12-30:
It is the Micro MinimOSD.Just connected 5V, GND TX and RX.
I will flash it again.

ShikOfTheRa, 2016-12-30:
Weird. Suggest start with config.h baud at 115k and gui at 115k to start. Thats default. Once all good maybe try them both at another baud

dan13373, 2016-12-30:
Will Don and report

dan13373, 2016-12-30:
Troubleshooting Steps:
1) Cleared EEPROM
2) In config changed BAUD Rate to 57k and 115k (before changing, started at 1)
3) Restarted Mac and unplugged USB Adapter
4) Opened GUI and changed comm settings

No Luck. Still says "Waiting". But the Upload in Arduino works all time. Is there something I miss?

ShikOfTheRa, 2016-12-30:
send email to gary@flypix.co.uk for next step...

jflyper, 2016-12-31:
That's weird...

Can you download the fresh scarab_osd dev package and start from download the sketch?

closed 2017-01-22 by ShikOfTheRa

brandonhill, 2017-09-18:
Having same issue..

kenw001, 2019-04-09:
Same issue here. Any solution update?
Thanks!