ShikOfTheRa/scarab-osd #348: Time Speeds Up

Nightsd01, 2016-12-30:
When I turn on my quad, after about 30-60 seconds, my MW OSD (1.6.5) has a strange issue. Suddenly, the timer speeds up. It will act as if it's constantly losing and regaining connection to my flight controller: it will show 0 satellites and then immediately show the real number (14), cycling between the two. It also shows 'No Data' but then it keeps updating with new data...

I suspected a hardware or wiring issue, so I tried using two separate minimOSD boards and they both experience exactly the same problem. I tried different cables, a different TELEM port on my pixhawk, and even a different pixhawk altogether.

In the Config.h file, it has the exact same problem regardless of whether I select #define PIXHAWK or #define APM. I followed all of the steps in the 'Mavlink Setup.md' documentation file.

Here is a video of the problem, it occurs in at 30 seconds:
https://youtu.be/7bt2he57hjw

jflyper, 2016-12-31:
So this is the APM problem... What is "time speed up"?

ShikOfTheRa, 2017-01-01:
@Nightsd01 are you using 1.6.5 from the releases tab ( https://github.com/ShikOfTheRa/scarab-osd/releases ) or is it the very latest from the main repository in last few days. 

I ask because there has been a bugfix since 1.6.5 release and that could cause symptoms described

ShikOfTheRa, 2017-01-05:
@Nightsd01 

Can someone try this. Today I have spent long time understanding mavlink protocol. Made a number of changes which I think may fix this. I have sent hundreds of thousands of mavlink packets form various sources and it seems stable and working now.

The only thing I am unsure of is armed status. The test data I have seems inconsistent with the documentation.

Needs to be version uploaded today:
https://github.com/ShikOfTheRa/scarab-osd/archive/master.zip

ShikOfTheRa, 2017-01-10:
@Nightsd01 are you working ok now with latest?

closed 2017-01-13 by ShikOfTheRa