ShikOfTheRa/scarab-osd #349: Cannot open GUI - Error missing "lib" folder

HughAV, 2016-12-31:
I am trying to edit the osd on my diatone sp3 vtx. runs mwosd. cannot get gui to open. message says "lib" folder missing. Is this not a part of the download?

jflyper, 2017-01-01:
Can you elaborate? Windows 7, 10, other? MacOSX? What's the MWOSD_GUI version? Screen shot of the message?

HughAV, 2017-01-01:
thank you for replying so fast! Windows 10, most recent download of MWOSD,
Github messenger would not let me post the snip. here it is:

[image: Inline image 1]

ShikOfTheRa, 2017-01-01:
have you tried 32 bit and 64 bit versions? Just as fyi, the 32/64 bit doesn't refer to windows you are using. Its anctually the jave component. I tried 32 bit and its working for me.

ShikOfTheRa, 2017-01-10:
@HughAV are you working now?

HughAV, 2017-01-10:
Thanks for following up. Have been busy, and have not had a chance to get
back in and try. I will let you know.

ShikOfTheRa, 2017-01-22:
Closing this for now. If have continued issues please raise on support forum. Thanks!

closed 2017-01-22 by ShikOfTheRa

HughAV, 2017-01-24:
Figured it out! Issue with my PC...

ShikOfTheRa, 2017-01-24:
OK - thanks for feedback