ShikOfTheRa/scarab-osd #350: Displayed flight modes

RCnator, 2017-01-01:
I use scarab-osd on my naze32 with iNav where I have 1 acro, 2 horizon, 3 baro (+2) , 4 heading (+3), 5 pos hold (+4), 6 rth and 7 waypoint. But on my osd it can only display ACRO/HORIZON/RTH/HOLD. Any plans on making it possible to display more types of flight modes? I am new to Github so I was unsure if I should leave this feedback here or elsewhere.

ShikOfTheRa, 2017-01-01:
In last few days I added:
AIRMODE
Waypoint

I havent tested but would love to know if the waypoint works. It should be like this, but I don't know if iNav is compatible with multiwii format yet. You are welcome to try!! 

https://youtu.be/FCIyhbT1kK0?t=8

Note
For option 3 will show horizon ICON - but you can enable the sensors to be displayed too (vertical  acc and mag igons)

RCnator, 2017-01-02:
Yea I added waypoint but I don't remember if it is currently actually usable. Might have been something relating to uploading of waypoints that wasn't finished.

Anyway, I enabled the sensors to be displayed but only acc shows up. Still I get altitude (baro and gps) readings and the compass works. Would have been great if this worked.

ShikOfTheRa, 2017-01-03:
I have just packaged up the latest release - 1.6.6.0

Have tested with iNav as far as I can with a simulator and after a few changes I think it will work now. I forgot it was only enabled for multiwii v2,4

Would be great if you can confirm. Likely to be a few weeks before I can verify in flight - just finishing off a quad to test

RCnator, 2017-01-04:
Oh I see now you're owner! You mean the OSD will display any flight mode the FC uses? When I use the simulator MAG and BAR light up depending on flight mode but it doesn't work for me IRL. I need to use a different computer to load the osd firmware from so I am sorry but I think I will have to wait. Maybe there has been even more updates when I get to it.

ShikOfTheRa, 2017-01-04:
Sim should show flight modes, but looking at the code in last few days, iNav uses different id's to say multiwii or cleanflight.

e.g. enabling airmode on simulator will send i.d's used by multiwii/cleanflight as default. As a result, airmode will show waypoint icon on MWOSD running inav now. 

Put another way - the simulator is not 100% accurate for all platforms. Unfortunately keeping it accurate for all different FC is a nightmare. I just don't have time. And simulator is primarily for developing.

Anyway - try it with 1.6.6 - I hope its working OK. It definitely wouldn't before as iNav uses differnet id indicator and MWOSD was only set to show it for MultiWii

RCnator, 2017-01-04:
Ok thanks!

ShikOfTheRa, 2017-01-10:
@RCnator are you working now?

RCnator, 2017-01-17:
Hi, yes I have been rather busy working and haven't updated the fw or anything. Appreciate the work you do though.

ShikOfTheRa, 2017-01-22:
Closing this as I believe resolved. It works on test sim for me. Please re-open if issue

closed 2017-01-22 by ShikOfTheRa