ShikOfTheRa/scarab-osd #351: Error after 15 seconds

sniperxnl, 2017-01-01:
I plug my fc into my computer, plug in my battery, run the gui, select my com port with the correct baud rate, and the osd connects, yay. BUT, after aproximately 15 seconds, I get an error message in the cmd tab reading this invalid checksum for command 119: 123 expected, got 3 119 139 {254 3 254 3 254 3 254 3 254 3 45 32 32 32 36 77 60 11 220 7 254 3 254 3 254 3 254 3 254 3 45 32 32 32 36 77 60 11 220 7 254 3 254 3 254 3 254 3 254 3 45 32 32 32 36 77 60 11 220 7 254 3 254 3 254 3 254 3 254 3 45 32 32 36 77 60 11 220 7 254 3 254 3 254 3 254 3 254 3 45 32 32 32 36 77 60 139 119 254 3 254 3 254 3 254 3 254 3 45 32 32 32 36 77 60 11 220 7 254 3 254 3 254 3 254 3 254 3 45 32 32 32 36 77 60 11 220 7 254} [3] þ þ þ þ þ - $M Üþ þ þ þ þ - $M Üþ þ þ þ þ - $M Üþ þ þ þ þ - $M Üþ þ þ þ þ - $M_?wþ þ þ þ þ - $M Üþ þ þ þ þ - $M Üþ This error will then repeat itself every second or so until I hit the close com button. I also cannot quickly write anything to the osd in the beginning 15 seconds as it will jump straight to this error which will then repeat.

ShikOfTheRa, 2017-01-01:
what osd hardware? Anything else connected at same time? That is usual cause

sniperxnl, 2017-01-01:
i use a piggyosd from furiousfpv

ShikOfTheRa, 2017-01-01:
WHen connecting GUI, are you disconnecting it from the FC? its mandatory...

sniperxnl, 2017-01-02:
i use the serial passtrough

ShikOfTheRa, 2017-01-03:
never used passthrough so not really sure what impact that has. Its normally a sign of corrupted serial. Can expect a few errors, but not regular like that. At some point I'll have a similar setup so may be able to help more. 
Have you tried furious?

ShikOfTheRa, 2017-01-22:
Any luck? Are you working now

sniperxnl, 2017-01-23:
No

Robinhuett, 2017-01-23:
I don't have this issue on my piggy. I'm also using serialpassthrough (on a kombini).
But I use the MWOSD version that it comes with. So if this is about the current branch ignore this.

ShikOfTheRa, 2017-01-23:
Someone else posted similar:
http://fpvlab.com/forums/showthread.php?50686-Cannot-update-mwosd-mini-through-FC

ShikOfTheRa, 2017-02-27:
I'm struggling to replicate - its working OK for me. I have no way forward with this...
I think if can go direct and prove it works OK to raise with teh configurator guys for whichever FC you are using

ShikOfTheRa, 2017-03-07:
@sniperxnl 
Would be interested to see if the new GUI works for you. It has some adaptive transmission config designed to work with programming through a flight controller. It might be completely unrelated to your issues but is worth a try:

https://github.com/ShikOfTheRa/scarab-osd/tree/master/MW_OSD_GUI