ShikOfTheRa/scarab-osd #353: Font Upload issue

advancetom, 2017-01-02:
Hey today im tested 1.6.5.0
And Works Fine Till i changed the Font to mguz or Else !
After restart it will start with the Standard Font
Im Used this in 1.6.4.0 and Works whats wrong?
Im use BF 3.0.1 :)

ShikOfTheRa, 2017-01-03:
Have you read the faq - there are a number of common reasons for this

advancetom, 2017-01-08:
I think After Version 1.6.4 i have this Problems Looks like he dont dont Write successful to the eeprom After restart?

ShikOfTheRa, 2017-01-09:
I have tried several times with different fonts and works every time for me. That is with latest packaged 1.6.6.1 and GUI

I also tried with 1.6.5.0 and seemed to work OK. 
Please check the FAQ carefully - that covers most of the issues. Especially the voltage for programming - as can read/write settings OK, but not fonts if voltage is low.

ShikOfTheRa, 2017-01-22:
No other reports and uable to replicate.  So closing this.
Also suggest to try the chrome GUI in latest release

closed 2017-01-22 by ShikOfTheRa