ShikOfTheRa/scarab-osd #356: BUG - bugfix for resetting of some Betaflight advanced parameters

logotron, 2017-01-07:
I try all versions from R1.6.4 up to R1.6.6.1 and in all of them after saving change in OSD next parameters in BetaFlight is reset to zero:
yaw_p_limit = 0
setpoint_relax_ratio = 0
dterm_setpoint_weight = 0
yaw_accel_limit =  0.000
accum_threshold = 0
yaw_accum_threshold = 0

After this changers Quad flip on ground; there's no way to fly!!

ShikOfTheRa, 2017-01-08:
OK - will look at. what version of betaflight

logotron, 2017-01-09:
I am running BF 3.1 and I notice problem on almost all pre-releases.
To be more interested, I sniffing COM port I there is next communication:
09/01/2017 20:35:28.422 [OSD] - 24 4D 3C 00 70 70 
09/01/2017 20:35:28.431  [FC] - 24 4D 3E 1E 70 01 01 02 02 02 02 03 03 14 04 04 04 0F 00 00 22 0E 35 19 21 53 32 32 64 28 00 00 05 05 05 4A

But I never found command where OSD asking FC about Advanced PID parameters (24 4D 3C 00 5E 5E)

ShikOfTheRa, 2017-01-09:
Have found something. Updated main repository. I can't test for couple of days, but you are welcome to!

I see betafight has a few more parameters - will see if can add these into the menu as well

jflyper, 2017-01-10:
@logotron 
You can use the CMS on BF3.1.

logotron, 2017-01-10:
@jflyper  
I already using CMS and this is absolutely amazing feature. You did fabulous work, my respect!
The problem is if someone by mistake goes into "normal" menu and save changes, Quadcopter become useless (happen to me a few times).

jflyper, 2017-01-10:
@logotron 
IC.

lichtl, 2017-01-10:
Maybe normal menu should be disabled when using CMS (not sure if this is already possible).

ShikOfTheRa, 2017-01-10:
@logotron 
I was in rush to go out last night and didnt look properly. Its weird - lots of stuff missing out of teh advanced pid - been through and as far as can tell without trying on a FC its looks OK on simulator.

Uploaded to repository

@jflyper 
I haven't yet tried betaflight. Hoping to put on a quad this weekend and test / tryout.

any chance of this in iNAV?

ShikOfTheRa, 2017-01-10:
@lichtl - lets see if can do it

jflyper, 2017-01-10:
@lichtl 
Well, that is not that easy.

- MWOSD doesn't know if FC is capable of talking displayport (CANVAS) MSP until it first receives the first displayport message.
- MWOSD still needs its own menu to adjust MWOSD specific parameters.

There is however, one thing we can do; disable FC-oriented menus if CANVAS is enabled (default for BETAFLIGHT now).

@ShikOfTheRa 
The CMS itself is already merged in iNav. IIRC, displayport was postponed to 1.6.

ShikOfTheRa, 2017-01-10:
That's brill. Look forward to trying.

Havent seen canvas running yet, but is it possible to minimise the probability of dual CMS running?:

in serial.ino:
    if (armed || fontMode || configMode)  >>>     if (armed || fontMode )

    case 0: // Enter / hold canvas mode
      configExit() ... 
      canvasMode = true;
      break;

in mwosd.ino:

if  (canvasMode = true) configMode=0;

logotron, 2017-01-10:
I test new update; OSD still do not requesting MSP_PID_ADVANCED from FC,
code 24 4D 30 11 5E newer went to FC.

Saving data with MSP_SET_PID_ADVANCED (code 24 4D 3C 11 5F) working, but with all values equal to 0.

One more detail, BF MSP_PID_ADVANCED have 19 bytes in total, OSD writing 17.
Definition inside BF is:
    case MSP_PID_ADVANCED:
        sbufWriteU16(dst, currentProfile->pidProfile.rollPitchItermIgnoreRate);
        sbufWriteU16(dst, currentProfile->pidProfile.yawItermIgnoreRate);
        sbufWriteU16(dst, currentProfile->pidProfile.yaw_p_limit);
        sbufWriteU8(dst, 0); // reserved
        sbufWriteU8(dst, currentProfile->pidProfile.vbatPidCompensation);
        sbufWriteU8(dst, currentProfile->pidProfile.setpointRelaxRatio);
        sbufWriteU8(dst, currentProfile->pidProfile.dtermSetpointWeight);
        sbufWriteU8(dst, 0); // reserved
        sbufWriteU8(dst, 0); // reserved
        sbufWriteU8(dst, 0); // reserved
        sbufWriteU16(dst, currentProfile->pidProfile.rateAccelLimit * 10);
        sbufWriteU16(dst, currentProfile->pidProfile.yawRateAccelLimit * 10);
        sbufWriteU8(dst, currentProfile->pidProfile.levelAngleLimit);
        sbufWriteU8(dst, currentProfile->pidProfile.levelSensitivity);
        break;

jflyper, 2017-01-10:
@ShikOfTheRa Thanks, I'll see if your proposed solution work.

ShikOfTheRa, 2017-01-10:
@jflyper - not looked at it too much but sure there is something me can do!

ShikOfTheRa, 2017-01-10:
@logotron - I've got betaflight running. See issue. just updated with hopefully last patch. Can save settings, power off and go back in and seem to have saved.

That said - I've looked at configurator for first time. I can't see any of those settings in there!
Is it required?

Also very confused. Says super rate in configurator, and there is serial msp for it, but in log it says its no longer used.

This all makes me think...
1 I should remove advanced tuning
2 use canvas mode

ShikOfTheRa, 2017-01-11:
OK - thought more about this last night.

Have removed all the Advanced PID code. It seems too liable to change between versions and I don't want to mess anything up. 
This should resolve all the issues created by saving zero values to advanced pid settings.

This is on the following basis:

1 - It is all available in forthcoming Betaflight release + current dev This is the way to go. Full credit to Jflyper.
2 - The RC tuning is now split across 2 pages and includes the RC yaw expo as part of basic settings which has been requested.
3 - We will look to get Betaflight CMS and MWOSD CMS to co-exist.

I consider this closed, but will keep this open for any discussions for a week or two.
A separate new one will be created for Betaflight / MWOSD CMS co-existence improvements

logotron, 2017-01-11:
I agree, CMS is definitely a good way forward.

What about an additional feature like ON SCREEN PIDS?
I see that you remove this feature from commitment for 1.7 version.
This feature really helps with tuning Quadcopters.
You can check on YouTube, a lot of positive feedback:  
https://www.youtube.com/watch?v=8mutKDVSKZs&t=11s

ShikOfTheRa, 2017-01-11:
Not sure that was much of a flight ?! 

Only removed the advanced PID - only these (which did not work until last night!):
"YAW P LIMIT";		
"YAW RATE ACC LIMIT";		
"RATE ACC LIMIT";

But added 
"RC YAW EXPO"

On screen PIDs is OK, but I am not sure how to detect when it is in in flight tuning mode.

logotron, 2017-01-11:
In flight tuning mode is described in this video:
https://www.youtube.com/watch?v=JpnCMLmxme0&t=404s

Everything happening inside Flight Controller.
OSD only need to display all PIDs values on the main screen during the flight and refresh values requesting MSP_PID in the same main loop where is MSP_STATUS, MSP_ATTITUDE,...

Pid tuning can be on sperate Layout/page which needs only a few additional parameters, PID values, Main Voltage, Main Current, Flight Mode,...
As additional nice to have feature will be if we can assign one Aux switch on transmitter where we can switch between OSD Layout with PIDs and without PIDs.

ShikOfTheRa, 2017-01-22:
Bug fixed - thanks for pointing it out. Amazed no-one noticed before.

In flight tuning PID is added to the roadmap for future (prob next) release. 

Closing this. Thanks!

closed 2017-01-22 by ShikOfTheRa