ShikOfTheRa/scarab-osd #361: Betaflight Configure OSD option

u8mydog, 2017-01-09:
Should I be able to configure the OSD from betaflight configurator? 

My goal is to enable smartaudio using softserial on the FC and use OSD joystick commands to adjust VTX. 

I've enabled the OSD option in betaflight configurator but the configurator OSD tab reports that it can not communicate to the OSD hardware. I can use mw_osd_gui to configure the OSD.

I've read [https://github.com/ShikOfTheRa/scarab-osd/blob/master/OTHER/DOCUMENTATION/Canvas_CMS.md](https://github.com/ShikOfTheRa/scarab-osd/blob/master/OTHER/DOCUMENTATION/Canvas_CMS.md) and do not see a 'CANVAS' option in betaflight.

FC: F3S AIO v1.0 (includes minimOSD hardware, UART1=OSD, UART2=S.Port, UART3=SBUS)
Betaflight: v3.1.0-RC5 target SPRACINGF3 (appears to define CMS)
Scarab-osd: v1.6.6.1 (disabled MULTIWII, defined BETAFLIGHT and CANVAS_SUPPORT)

jflyper, 2017-01-10:
@u8mydog 
> I've enabled the OSD option in betaflight configurator but the configurator OSD tab reports that it can not communicate to the OSD hardware.

OSD feature selector on Betaflight configurator's Configuration tab is for FC integrated OSDs.

> I've read https://github.com/ShikOfTheRa/scarab-osd/blob/master/OTHER/DOCUMENTATION/Canvas_CMS.md and do not see a 'CANVAS' option in betaflight.
CANVAS is the MWOSD side build option.

> FC: F3S AIO v1.0 (includes minimOSD hardware, UART1=OSD, UART2=S.Port, UART3=SBUS)
Betaflight: v3.1.0-RC5 target SPRACINGF3 (appears to define CMS)
Scarab-osd: v1.6.6.1 (disabled MULTIWII, defined BETAFLIGHT and CANVAS_SUPPORT)

Your configuration looks okay.

Try "YAW Left" instead of "YAW Right" in the stick combination to enter CMS. ("YAW Right will take you to the MWOSD's menu).

u8mydog, 2017-01-22:
Thanks. I misunderstood the purpose of OSD option in betaflight. Verified that CMS works with yaw left, throt middle and elev stick up.

closed 2017-01-22 by u8mydog

MathijsG, 2018-04-08:
With Betaflight 3.3 this DOES work. But I have some fonts that are wrong, like he symbol for throttle value. How can I solve this?

Markuzzzz, 2018-05-27:
having the same problem... letters are ok only signs are wrong.