ShikOfTheRa/scarab-osd #362: Naze32 LTM altitude problem

tednv, 2017-01-11:
This happens when osd is turned on before gps has a fix. If gps gets fix and I power cycle osd, everything works as it should. If osd is turned on before gps fix, when disarmed altitude shows up correctly. When you get gps fix and you arm, altitude gets set to -2150 (in my location, I'm sure it's different for everyone else). When you fly, it does go up and down normally but relative to this 2150 value not 0. The value has been consistent for me every time, so I'm guessing it has to do with some offset either from the gps or the baro sensor in my flight controller (naze32 advanced). I tried moving some code around in LTM.h but what worked for me was commenting out line 38 as follows:

//    mw_ltm.GPS_altitude_home = GPS_altitude;

After this altitude shows correctly both in armed and disarmed state. I'm not sure what this is breaking (if anything) but it seems to have fixed the altitude problem for me

ShikOfTheRa, 2017-01-12:
is the OSD connected as normal - so can use menus etc. when not armed, then flips into LTM when armed?

tednv, 2017-01-12:
yes msp side works as expected when not armed, shows gps counts, can edit things, and altitude sets to zero correctly. when arming, altitude switches to -2150 if osd was powered on before a gps fix. if i leave osd powered off until there is a gps fix and then power it, everything works and altitude in armed matches disarmed. it does go up from -2150 when flying, so it's just not zeroing correctly. all other values when armed are correct and updating. commenting out that one line fixes it and armed and disarmed match

ShikOfTheRa, 2017-01-12:
I thought that was the case. In brief they are conflicting... 

So your suggestion will work for you but unfortunately not others who are using LTM without MSP.  If anyone uses mavlink and msp at same time, they may get same issue (I suppose no one do this?) 

I have submitted a PR to hopefully work in both scenarios. Care to test?

Thanks for pointing out. I don't suppose many use this combination. 

Just tried iNav - very impressed

tednv, 2017-01-13:
there will be a whole lot more people using it if i ever put up some more info on my build, i think i'm way ahead of most builds out there. i'm squeezing every ounce of processing power out of my skyline32 with inav (using custom inav builds)

oh also a heads up, the current doesn't come through ltm right, but i don't care just ended up soldering the current sensor wire that goes to FC to minimosd and now everything is working real good

i have RTH working, waypoints are working flawlessly, current sensor, bluetooth, blheli 4 way through bluetooth, and now osd MSP+LTM. my f3 board should be coming this week, the only reason i got it was to have more serial ports, i plan to add surface/sonar and led strip to list above. i have larger builds too but this one outperforms them all, and it's much easier to just throw in the back of the car. had led strip working and loved it but at the expense of the two softserials, ran out of timers too

will test your update and let you know if it works

tednv, 2017-01-13:
tested last commit, that took care of it!

ShikOfTheRa, 2017-01-13:
That's great setup there. On mine I use telemetry over audio to a tracker unit. Purely for range / signal quality. 

let me know if see any others.

The current is an issue - from what I read, some FC sent maH consumed and others send mA might have to have a #define for both options.

Thanks for feedback and raising issue. I'll close thsi for now. Feel free to re-open. I still get comments when closed though..

closed 2017-01-13 by ShikOfTheRa