ShikOfTheRa/scarab-osd #364: Betaflight / MWOSD CMS co-existance

ShikOfTheRa, 2017-01-11:
@jflyper 

Jflyper has added CMS support to some FC such as betaflight. 
This provides ability to access / configure all settings.

They can potentially clash in certain scenarios. This issue is to discuss and propose solutions. 

Potential options:

1:
in serial.ino:

if (armed || fontMode || configMode)  // change to: 
if (armed || fontMode )

case 0: // Enter / hold canvas mode
  if (configMode)
    configExit() // add this line to exit MWOSD config menu
  canvasMode = true;
  break;

not sure if also need, in mwosd.ino:

if (canvasMode = true) configMode=0;

Question - is it possible to initiate canvas mode from OSD? that would be great to add into MWOSD menu

ShikOfTheRa, 2017-01-11:
Actually I think just before the serial canvas case statement would work better. I'll submit a PR and try out tonight.

ShikOfTheRa, 2017-01-11:
Er. dull question - how do you enter CMS mode on the FC ?! I can't find...

skaman82, 2017-01-11:
if you mean the stick combo: yaw left+thr mid and pitch up

ShikOfTheRa, 2017-01-11:
Brill. Have updated the main repository. Can't wait to try this out tonight!

ShikOfTheRa, 2017-01-11:
Do you have to do soemthing to enable it - or is only ther ein F3 targets? Doesn't seem to obe initiation on my naze running bf 3.1

skaman82, 2017-01-11:
Dunno about F1 targets I just needed to enable canvas in MWOSD and then it worked perfectly. Have no F1 target anymore to test...

jflyper, 2017-01-11:
Ah, F1 targets drops CMS. Must be >128K targets.

ShikOfTheRa, 2017-01-11:
OK. i do now have an f3 so will have to get that running. Was hoping to see in action tonight.
Any chance someone can try latest in repo - to see if the two play ball together now? Had a look at the code and it looks like it should

u8mydog, 2017-01-12:
There is not enough memory to enable CMS but if you disabled some functionality like BARO and LED_STRIP in src/main/target/NAZE/target.h, then it will link and appears to fit in memory. Compile beta flight yourself using the command:
make TARGET=NAZE OPTIONS=CMS

ShikOfTheRa, 2017-01-12:
OK - did bits in post 1 - works fantastic. 
Tried all sorts of combos to get to lock / cause issues. Couldn't break it. 

@jflyper any need to retain the canvas first / * ? 
Have left it in but it seems to work great. Bit odd having different stick commands. I guess they were there because of co-existance issues. 

You did a great job on this!

jflyper, 2017-01-17:
@ShikOfTheRa 
The upper left asterisk is to indicate that there is MSP_DISPLAYPORT message coming in without starting. Can delete it now, but leaving it might help trouble shoot some cases.

ShikOfTheRa, 2017-01-17:
Yeah its all merged and tested. I can say it works awesome.
I also put an extra menu page in for betaflight 3.1 to access or "advertise" it, because I think it's a largely unknown feature.

really would like to see this in iNav - fantastic for easy  field tuning so have just added a note re that

ShikOfTheRa, 2017-01-17:
Oh - no issues in leaving the * - in testing I couldn't make it occur any more. I guess that's a positive

jflyper, 2017-01-17:
@ShikOfTheRa Awesome!!!

ShikOfTheRa, 2017-01-22:
Think this can be closed. Feedback v positive.

closed 2017-01-22 by ShikOfTheRa