ShikOfTheRa/scarab-osd #367: list of supported software

mahowik, 2017-01-16:
you can also include my fork to the list of supported software
https://github.com/mahowik/mahowii

ShikOfTheRa, 2017-01-17:
Wow. There’s a blast from the past!! MultiWii and mahowik!

Some interesting developments there. Some of those vids showed some excellent position hold.

I don’t suppose this would fit on a 328?? 
Only have one multiwii copter. I’m resisting updating it!!

Have updated the multiwii line and included url to your repo. Hopefully it will make a few other look.

mahowik, 2017-01-17:
Thanks a lot.

With 328 it's possible to fit but without GPS.

Sorry but I didn't found the description update anywhere. Did you forgot to commit? :)

ShikOfTheRa, 2017-01-17:
I put in config.h like this

//#define MULTIWII                  // Uncomment this if you are using latest 2.4 MULTIWII / MAHOWII  (https://github.com/mahowik/mahowii)

I thought so. 
I'll have a look. I do have one of these.. http://www.multiwiicopter.com/products/paris-v5r3-multiwii-with-full-osd-gps-mag-board

mahowik, 2017-01-18:
thanks a lot

closed 2017-01-22 by ShikOfTheRa