ShikOfTheRa/scarab-osd #370: Piggy OSD

sniperxnl, 2017-01-18:
Hello All,

Can i load the latest FW from scarab-osd

Define Canvas and Define latest BF version
Can i then set the settings with BF OSD menu option in the GUI?

ShikOfTheRa, 2017-01-18:
I don't know. Suggest you raise with furiousfpv. 

The guys are getting furious with their lack of response .... bad joke eh! 

I have one, but scared to try it in case I screw it up!

sniperxnl, 2017-01-18:
furiousfpv respons always on the emails what i have send to them.  this have nothing to do emwith furiousfpv so. 

i have loaded the latest firmware todat and define canvas.  only i can't use the osd features in betaflight gui to set everything up

ShikOfTheRa, 2017-01-18:
We are / were unsure about the piggy hardware - Furious haven't confirmed its compatible.
But if you say it works, that's great. I'll upgrade mine now.

For canvas features...
Must have Betaflight 3.1 (at moment it is only available as unstable release)
Must have a build that includes the dispaly port code (I think F3 processor minimum)

OSD features in betaflight are not for MWOSD

ShikOfTheRa, 2017-01-22:
Confirmed standard firmware. This can be closed.

Yes MWOSD supports piggy as standard

closed 2017-01-22 by ShikOfTheRa