ShikOfTheRa/scarab-osd #371: 1.6.7 Canvas mode (MSP_DISPLAYPORT) going OoS

jflyper, 2017-01-29:
There has been multiple reports of 1.6.7 showing asterisk with screen cleared upon entry into canvas mode.

https://www.rcgroups.com/forums/showpost.php?p=36754674&postcount=44074

jflyper, 2017-01-29:
Strange that no MSP_DISPLAYPORT message seems to be understood after displaying the asterisk.

ShikOfTheRa, 2017-01-29:
My error I think. Cleaning up compiler errors for unused variables...
Just undone...

jflyper, 2017-01-29:
It actually was my fault to make this an option. I should have coded the attribute field as mandatory on both sides!!!

ShikOfTheRa, 2017-01-30:
No - definitely my fault. I haven't tested to confirm. Will do this week. Could do with a copter of each type to verify everything without constantly rebuilding...

closed 2017-02-12 by jflyper