ShikOfTheRa/scarab-osd #372: can not navigate in menu

robschii, 2017-01-31:
i use microminim osd from banggood  and i use the last beta.

i can open the menu  mitch trottle stick middle and pitch foward  an also exit  but i can not change the  page   witch yaw right and trottle middle    like in the release 1.6

ShikOfTheRa, 2017-02-01:
That's strange - its working for me 

values sent from TX must go outside these values (which can amend in globalvariables:
#define MAXSTICK         1850
#define MINSTICK         1150

They are quite high, but in line with how shoudl be configured with a normal FC.
Also worth checking that rates are not enabled.

Odd thing is can enter the menu.... which uses same values!

ShikOfTheRa, 2017-02-03:
Just to be sure - what controller do you have set in config.h

closed 2017-02-20 by ShikOfTheRa