ShikOfTheRa/scarab-osd #373: Add mAh/min indicator

Robinhuett, 2017-02-01:
Hi,

I'd love to see an mAh/minute indicator in the OSD. Kinda linke in the ImmersionRC and TBS OSDs.
I'll have a look in the code myself. 
Maybe I can figure out how to implement it, than I'll make a PR.

Tanks in advance.

ShikOfTheRa, 2017-02-01:
Suggest implementing in method similar to this. I haven't seen either of those so not quite sure how operate, but probably these have all the basics of what is needed to impelement- :

#define DISPLAYWATTS                // Enable this to display Watts (if Watts selected in layouts). Select either WATTS or EFFICIENCY. Not both
//#define DISPLAYEFFICIENCY         // Enable this to display Watts/KMh or Mph for efficiency (if Watts selected in layouts). Select either WATTS or EFFICIENCY. Not both

What do they measure - is it instantaneous, average over fliy time or measured over last minute?

Robinhuett, 2017-02-01:
This is an ongoing calulation (average over flighttime), so im not entirly sure if it would have an large impact on the performance. But since this isn't a critical value a slightly slower calculation would be more than enough.
So it would be simple math like: **mAh used** / **time**.

ShikOfTheRa, 2017-02-01:
Plenty of processing power available. Memory free isnt huge, but this is quite small. Need to think carefully about what you are asking...
Have just gone to implement and  I think you are probably asking for mah used per minute to determine how many minutes left of flying time??

Anyway, have tried to implement something. Hope its what you want
Its based upon flying time so will take a minute or two to settle to a more accurate figure.

Robinhuett, 2017-02-01:
Wow, that was fast!
From what I see in the commits it looks good 👍 

I haven't made anything yet. I could start when I'm at home. But I guess I could simply test yours :)

ShikOfTheRa, 2017-02-01:
Yeah - try this. It works on simulator.... but not sure if its what you are looking for.

Robinhuett, 2017-02-06:
Seemd to work fine on the bench. I'll test it in the field and will check the math when the wether gets better in germany..
But for now seems good. If there is something to change ill make a PR ;)

closed 2017-02-06 by Robinhuett