ShikOfTheRa/scarab-osd #377: xjt transmitter telemetry, Will not display osd over video.

triolus, 2017-02-11:
No option to take xjt transmitter sport telemetry to osd. Osd will not overlay over video. when video is off the osd data is available.

jflyper, 2017-02-12:
@triolus 
Are you talking about receiver side OSD?

ShikOfTheRa, 2017-02-13:
I think so. Not to bad to do, but I suspect requires hardware inverted if its like rest of FRSKY...

I don't have any plans as no time. Added to development wishlist if anyone else can pick this up

triolus, 2017-02-20:
Yes. I do not know if the FRSKY xjt is set up to just transmit proper fsky telemetry data or if it is only designed for their own monitor. I made a small error and will retest;I was having the OSD transmit go back to the XJT. It gives an option for which hookup: uart or frsky, and will test the uart option without transmitting back.

ShikOfTheRa, 2017-02-27:
I am closing this to keep the outstanding issues list clean.
It is captured on the list of requests should anyone want to pick this up.
I did some googling and there do appear to be some other OSD options out there for you.

closed 2017-02-27 by ShikOfTheRa