ShikOfTheRa/scarab-osd #379: PASSTHRU mode for fixed-wing not being detected for iNav 1.6

csurf, 2017-02-12:
I'm having an issue with MWOSD not being able to detect when the flight mode is PASSTHRU for the latest version of iNav 1.6. The flight mode still shows as being in 'GYRO' mode when it should show 'PASS'.

I tried looking at both the MWOSD code and the iNav code, and couldn't figure out why this is happening. Nothing major appears to have changed in either the BOXID or MSP_STATUS code for both iNav and MWOSD with respect to being able to read the PASSTHRU mode bit flag from the MSP_STATUS command.

ShikOfTheRa, 2017-02-13:
I just tried with simulator of latest test release - 1.6.8 and it worked OK. 
inav appears to be using same value, so would expect this to work on other versions too.
note that mission and airmode are different to other FC, so need to use 1.6.8 if want those to be accurate

i'll test when get it on a plane, but likely to be 2-3 weeks

csurf, 2017-02-14:
I looked at the code for both iNav and MWOSD, and, I agree, the code looks the same, but mode indicator is still showing GYRO when in PASSTHRU. All other modes appear to work fine. I'm not using mission or airmode. 

Any other ideas?
I'd be willing to test if u can send me a patch or something to work with...

ShikOfTheRa, 2017-02-14:
Grab zip from download link here:
https://github.com/ShikOfTheRa/scarab-osd

Find and enable DEBUG 4 in config.h

On the debug screen displayed on OSD,
D0 – Sensor ID Active status – first 16 
D1 – Sensor ID for camstab (as a secondary reference)
D2 – Sensor ID bit that MWOSD thinks is used for passthru – so 2048 would be bit 11

In example above, when toggle passthru mode, should see bit 11 toggle.

Bits above 15 are ignored in this test

ShikOfTheRa, 2017-02-14:
Also, do you have one of following enabled::
MIXER_FLYING_WING
MIXER_AIRPLANE
MIXER_CUSTOM_AIRPLANE;

csurf, 2017-02-14:
MIXER_FLYING_WING

ShikOfTheRa, 2017-02-14:
OK that's good. It must be one of those or it doesn't send. 
Have had a look through both sides and appear Ok to me .

ShikOfTheRa, 2017-02-14:
I just tried with mwosd 1.6.8 and inav latest dev release and it worked ok.  simple setup with a naze. only enabled horizon and passthrough for test. One othe rperson has reported on rcgroups
Are you using LTM or mavlink?

csurf, 2017-02-15:
ok, tried enabling debug, and I'm getting all zeros for D0 - D3. What am I doing wrong?

ShikOfTheRa, 2017-02-15:
Yes that helps. (sorry I forgot to push the last commit... so ya wouldn't have seen any change!)

Quiet alot of additional features added into inav and I think this is pushing the send order of id's outside of the original multiwii scope built. 
Passthrough with fixedwing enabled is first one to hit this limit. 
MWOSD needs to be amended to support great new features inav is bringing

Have amended to support and allow extra growth most likely in the GPS area. Can you try latest I have just uploaded now?

ShikOfTheRa, 2017-02-15:
If already downloaded,  I left in error from testing so if no icon is shown, please comment out NAZAMODECONTROL in config.h

ShikOfTheRa, 2017-02-15:
See for hopefully a resolution

https://github.com/ShikOfTheRa/scarab-osd/issues/379

csurf, 2017-02-15:
@ShikOfTheRa,
ok, adding the new modes to the list fixed it.

Before I close this out, I'm curious to know how the ordering & logical bit math works. Is the 'mode' struct some kind of bit-packed structure? Why did adding the new modes into the struct fix the problem? Wouldn't the 'switch-case' within Serial.ino automatically take care of skipping the unused modes?

ShikOfTheRa, 2017-02-16:
Excellent - thanks for feedback.

There are three parts to this.

Index
Each possible feature that is sent has a unique id. In iNav, Passthrough = 12

Boxid:
Because there are many potential features that can be enabled in code – and from historic lack of memory available with multiwii / atmaga 328 from which todays FC came from, only ones available for use are sent. 
So boxid sends the order of features available as a stream of bytes.
In your inav, Passthrough “id” 12 is sent in byte 16.
This only has to be sent/requested once. It never changes.

MWsensorsActive
When a feature is enabled, it toggles a bit in MWsensorsActive and sends in MSP_IDENT
On OSD side it reads boxid and see a match for the id of 12 in byte 16. 
So now OSD knows that bit 16 in the SensorsActive (within MSP_IDENT) field corresponds to the Passthrough bit.

Back to the issue, again for reducing memory on old 328 processors, the lower order bytes received structure is stored as 8 or 16 bits. Passthrough is sent later in the stream compared to previous as many new modes added into iNav. This shifted it from out of the previous OK 16 bit section.

Its MWOSD issue by design to work with small memory on 328. Should be good now to end of life!

ShikOfTheRa, 2017-02-20:
i think can be closed - re-open if necessary.

As fyi, i will add more modes in soon.

closed 2017-02-20 by ShikOfTheRa